rvdbreemen/OTGW-firmware #1: I think it is that i search for!

vampywiz17, 2020-12-08:
Hello there, Now i use OTGW with espeasy + serial server. If i read correctly, this FW is able to get a more sophisticated method (send datas with MQTT)

It sound good to me!

But i would like to ask some questions.

 My NodeMCU is work really strange with espeasy, when send data with ser2net, the board is not answering from the ping and the web interface is not accessable some sec. I think the reason that send to much data same time with the serial and it is to much  the NodeMCU. i hope with your firmware is not happen bacuse with MQTT (like tasmota) i never experience this.

rvdbreemen, 2020-12-08:
I have been using the firmware as it is for a couple of weeks now, and it is stable and capable of sending MQTT data reliable. I have it integrated with Home Assistant, but you could use another platform as your home automation and it should work fine. 
The whole reason why I started this firmware is exactly what you experience with easyesp, beyond the fact that I like tinkering with IoT and creating a firmware from scratch of course ;-)

Just let me know what you think if you do try it, and any suggestions are more than welcome ofcourse.

DaveDavenport, 2020-12-08:
I have been using this firmware too for the past weeks and it has been working reliable for me.

DaveDavenport, 2020-12-08:
Thanks for putting this online.  It saved me from having to write this myself :D .

I have some things I want to do in future, have not looked at this yet (make a merge request if it is not (yet) possible):

- [ ] Update time on thermostat. (if possible)
- [ ] Set outside temperature via mqtt. 
- [ ] Set setpoint via mqtt.

(sorry for hijacking this post, are you on irc or other channel?)

rvdbreemen, 2020-12-08:
@DaveDavenport no problem. I have just not have enough time to continue my development. You can find me on https://gitter.im/ to talk on the project.

rvdbreemen, 2021-01-03:
> Thanks for putting this online. It saved me from having to write this myself :D .
> 
> I have some things I want to do in future, have not looked at this yet (I might make a merge request if it is not (yet) possible):
> 
> * [ ]  Update time on thermostat. (if possible)
> * [ ]  Set outside temperature via mqtt.
> * [ ]  Set setpoint via mqtt.
> 
> (sorry for hijacking this post, are you on irc or other channel?)

Ever since version 0.4.x

rvdbreemen, 2021-01-03:
Sending commands can done since version 0.4.1

closed 2021-01-03 by rvdbreemen