rvdbreemen/OTGW-firmware #4: Webpages setting page stay's empty.

Marcello1970, 2020-12-21:
The software is working, I see the values in in the MQTT broker. About one thing i'm not clear, the webpages: 
What can I do with the the web pages? The main pages shows 'Wait! retrieving local messages .....' The settings page shows noting. There is a setting.ini file created, but it only contains the host name. Only the 'FSexplorer ESP' works. Is that right?

rvdbreemen, 2020-12-21:
That correct. The settings stuff and the webpages / UI still need to be done, these are the defaults. I am considering to use ESP Dash v3. But that needs some refactoring to an async library set. But once that's done the UI should be lovely too ;-)

rvdbreemen, 2020-12-24:
It's a work in progress. The settings code and UI are still not done. Keep tracking the project it will happen.

closed 2020-12-24 by rvdbreemen

rvdbreemen, 2021-01-03:
@Marcello1970 the setting page is done.  Echt up some sort of UI.