rvdbreemen/OTGW-firmware #5: Implement all known OpenTherm Data-ID's.

STemplar, 2021-01-07:
Hi,

Is it possible to implement all known OpenTherm Data-ID's? I have a Brink Renovent Excellent 300 HRU/WTW which supports OpenTherm and the ability of parsing ventilation / heat-recovery Data-ID's would be nice.

Greetings,

STemplar

rvdbreemen, 2021-01-07:
Love to add those data-id's do you have information on them for me? Because I have implemented all the known Data-ids from the spec I have seen, which is version 2.2. Do you have any documentation on the data-ids for me?

rvdbreemen, 2021-01-09:
@STemplar do you know the data-id's we are talking about. I have implemented documented [in the specification 2.2 (link)](https://github.com/rvdbreemen/OTGW-firmware/blob/main/Specification/Opentherm%20Protocol%20v2-2.pdf). Any guidance on the data-id's to be implemented would be needed. And would you be willing to test it for me?

STemplar, 2021-01-12:
Okay, here is the info.

Data-Id Map.otm file from the OpenTherm Testtool v24 (OpenTherm Technical Specifications v2.3b) 
```
;Data-Id Map (OpenTherm Technical Specifications v2.3b)
;Remarks: This is a complete list of available defined ID's
;	it can be edited with a text editor. combined read/write ID's
;	are on 2 seperate lines. To reduce the number of lines,
;	delete non used ID's

0,STATUS,READ,FLAG,00000000,FLAG,00000000,Yes
1,"CONTROL SETPOINT",WRITE,F8.8,0,100,"10,00",Yes
2,"MASTER CONFIG/MEMBERID",WRITE,FLAG,00000000,U8,0,255,0,Yes
3,"SLAVE CONFIG/MEMBERID",READ,FLAG,00000000,U8,0,255,0,Yes
4,"COMMAND",WRITE,U8,0,255,2,U8,0,255,0,Yes
5,"FAULT FLAGS/CODE",READ,FLAG,00000000,U8,0,255,0,Yes
6,"REMOTE PARAMETER SETTINGS",READ,FLAG,00000000,FLAG,0000000,Yes
7,"COOLING CONTROL",WRITE,F8.8,0,100,"0,00",Yes
8,"TsetCH2",WRITE,F8.8,0,100,"10,00",Yes
9,"REMOTE ROOM SETPOINT",READ,F8.8,-40,127,"0,00",Yes
10,"TSP NUMBER",READ,U8,0,255,0,U8,0,0,0,Yes
11,"TSP ENTRY",READ,U8,0,255,0,U8,0,255,0,Yes
11,"TSP ENTRY",WRITE,U8,0,255,0,U8,0,255,0,No
12,"FAULT BUFFER SIZE",READ,U8,0,255,0,U8,0,0,0,Yes
13,"FAULT BUFFER ENTRY",READ,U8,0,255,0,U8,0,255,0,Yes
14,"CAPACITY SETTING",WRITE,F8.8,0,100,"0,00",Yes
15,"MAX CAPACITY / MIN-MOD-LEVEL",READ,U8,0,255,0,U8,0,100,0,Yes
16,"ROOM SETPOINT",WRITE,F8.8,-40,127,"0,00",Yes
17,"RELATIVE MODULATION LEVEL",READ,F8.8,0,100,"0,00",Yes
18,"CH WATER PRESSURE",READ,F8.8,0,5,"0,00",Yes
19,"DHW FLOW RATE",READ,F8.8,0,16,"0,00",Yes
20,"DAY - TIME",READ,U8,0,255,0,U8,0,59,0,Yes
20,"DAY - TIME",WRITE,U8,0,255,0,U8,0,59,0,No
21,"DATE",READ,U8,1,12,1,U8,1,31,1,Yes
21,"DATE",WRITE,U8,1,12,1,U8,1,31,1,No
22,"YEAR",READ,U16,1900,2099,2002,Yes
22,"YEAR",WRITE,U16,1900,2099,2002,No
23,"SECOND ROOM SETPOINT",WITE,F8.8,-40,127,"0,00",Yes
24,"ROOM TEMPERATURE",WRITE,F8.8,-40,127,"20,00",Yes
25,"BOILER WATER TEMP.",READ,F8.8,-40,127,"20,00",Yes
26,"DHW TEMPERATURE",READ,F8.8,-40,127,"20,00",Yes
27,"OUTSIDE TEMPERATURE",READ,F8.8,-40,127,"10,00",Yes
28,"RETURN WATER TEMPERATURE",READ,F8.8,-40,127,"19,00",Yes
29,"SOLAR STORAGE TEMPERATURE",READ,F8.8,-40,127,"0,00",Yes
30,"SOLAR COLLECTOR TEMPERATURE",READ,F8.8,-40,127,"0,00",Yes
31,"SECOND BOILER WATER TEMP.",READ,F8.8,-40,127,"20,00",Yes
32,"SECOND DHW TEMPERATURE",READ,F8.8,-40,127,"20,00",Yes
32,"EXHAUST TEMPERATURE",READ,S16,-40,127,20,Yes
48,"DHW SETPOINT BOUNDS",READ,S8,0,127,0,S8,0,127,0,Yes
49,"MAX CH SETPOINT BOUNDS",READ,S8,0,127,10,S8,0,127,90,Yes
50,"OTC HC-RATIO BOUNDS",READ,S8,0,40,0,S8,0,40,0,Yes
56,"DHW SETPOINT",READ,F8.8,0,127,"10,00",Yes
56,"DHW SETPOINT",WRITE,F8.8,0,127,"10,00",No
57,"MAX CH WATER SETPOINT",READ,F8.8,0,127,"90,00",Yes
57,"MAX CH WATER SETPOINT",WRITE,F8.8,0,127,"90,00",No
58,"OTC HEATCURVE RATIO",READ,F8.8,0,40,"0,00",Yes
58,"OTC HEATCURVE RATIO",WRITE,F8.8,0,40,"0,00",No

; New ID for ventilation/heat-recovery applications

70,"STATUS V/H",READ,FLAG,00000000,FLAG,00000000,Yes
71,"CONTROL SETPOINT V/H",WRITE,U8,0,100,0,Yes
72,"FAULT FLAGS/CODE V/H",READ,FLAG,00000000,U8,0,255,0,Yes
73,"DIAGNOSTIC CODE V/H",READ,U16,0,65000,0,Yes
74,"CONFIG/MEMBERID V/H",READ,FLAG,00000000,U8,0,255,0,Yes
75,"OPENTHERM VERSION V/H",READ,F8.8,0,127,"2,32",Yes
76,"VERSION & TYPE V/H",READ,U8,0,255,1,U8,0,255,0,Yes
77,"RELATIVE VENTILATION",READ,U8,0,255,0,Yes
78,"RELATIVE HUMIDITY",READ,U8,0,255,0,Yes
78,"RELATIVE HUMIDITY",WRITE,U8,0,255,0,No
[... 176 lines cut ...]
ID113: Number of un-successful burner starts
ID114: Number of times flame signal was too low
ID115: OEM-specific diagnostic/service code
ID116: Number of succesful starts burner
ID117: Number of starts CH pump
ID118: Number of starts DHW pump/valve
ID119: Number of starts burner during DHW mode
ID120: Number of hours that burner is in operation (i.e. flame on)
ID121: Number of hours that CH pump has been running
ID122: Number of hours that DHW pump has been running or DHW valve has been opened
ID123: Number of hours that burner is in operation during DHW mode
ID124: The implemented version of the OpenTherm Protocol Specification in the master
ID125: The implemented version of the OpenTherm Protocol Specification in the slave
ID126: Master product version number and type
ID127: Slave product version number and type
```

Reverse engineered Remeha Data-Id's (only used/valid when MemberID's of 11 is returned with Data-ID 3.
```
ID 131:   {u8 u8}   "Remeha dF-/dU-codes"
ID 132:   {u8 u8}   "Remeha Servicemessage"
ID 133:   {u8 u8}   "Remeha detection connected SCU’s"
```

More info: [https://www.domoticaforum.eu/viewtopic.php?f=70&t=10893](https://www.domoticaforum.eu/viewtopic.php?f=70&t=10893)

rvdbreemen, 2021-01-19:
@STemplar thank you so much for pointing me in the right direction. I will look at implementing this ASAP. Should not be to hard. Unless information is overlapping with existing MsgID's.

STemplar, 2021-01-19:
> 
> 
> @STemplar thank you so much for pointing me in the right direction. I will look at implementing this ASAP. Should not be to hard. Unless information is overlapping with existing MsgID's.

That should not be the case. The only trick is dealing with those Remeha Data-ID's in the range 128 - 255 if you are implementing them. Those Data-ID definitions are only valid if the Member-ID of 11 is returned with Data-ID 3 i.e. the boiler identified itself as a Remeha device.

rvdbreemen, 2021-01-19:
@STemplar can you test the H/V msgids for me to validate function?

STemplar, 2021-01-20:
> 
> 
> @STemplar thank you so much for pointing me in the right direction. I will look at implementing this ASAP. Should not be to hard. Unless information is overlapping with existing MsgID's.

FYI: Found this OpenTherm Testtool link, it is still active: [http://www.hccp.nl/downloads/OpenThermMonitor.zip](http://www.hccp.nl/downloads/OpenThermMonitor.zip).

rvdbreemen, 2021-01-20:
@STemplar thanks for the tool. I downloaded it, it contains interesting documentation. FYI I just pushed the implementatoin of all other msgid's to the dev branch. I will publish a binary release as soon as I am able to test it myself. #weekend

STemplar, 2021-01-20:
> 
> 
> @STemplar can you test the H/V msgids for me to validate function? Because in the dev branch I just implemented the IDs 70-91 for H/V messages.

Hi rvdbreemen,

I want to but can I use the binary for my setup or do I need to compile my own? I've got a slightly differend hardware setup. The classic OTGW hardware with the Recom DC/DC convertor mod. And instead of the MAX232 IC, I connected a ESP8266-12E VCC=5V. On it I run the ESP-LINK: Wifi-Serial Bridge w/REST&MQTT V2.2.3. To use your Nodemcu firmware the only thing I need to do is altering these PIN definitions in the OTGW-Core.ino file? It has been 4 years ago I set this up.

```
#define PIN_I2C_SDA 4
#define PIN_I2C_SCL 5
```

rvdbreemen, 2021-01-20:
@STemplar I did not realize you are not on the NodoShop version. The PIN definition is currently a "defined" thing. So you have to compile it yourself. What I could do is make the pin configurable. That way it could become a simple setting that can be set thru the settings file.

STemplar, 2021-01-20:
> 
> 
> @STemplar I did not realize you are not on the NodoShop version. The PIN definition is currently a "defined" thing. So you have to compile it yourself. What I could do is make the pin configurable. That way it could become a simple setting that can be set thru the settings file.
> 
> I can even compile it for you. What you like.

First I have to check if the PINs are indeed different. The Nodemcu is outfitted with the same ESP-12 module if I am not misstaken.

STemplar, 2021-01-20:
I got this unit: [https://www.dx.com/p/esp-12-esp8266-serial-wi-fi-wireless-module-w-built-in-antenna-compatible-3-3v-5v-for-arduino-2057735.html](https://www.dx.com/p/esp-12-esp8266-serial-wi-fi-wireless-module-w-built-in-antenna-compatible-3-3v-5v-for-arduino-2057735.html)

flash vendor: E0h : N/A
flash devID:4014h
QUAD;8Mbit
crystal:26 Mhz

Flash map 4MB:512/512, manuf 0xE0 chip 0x4014

Flashing to OTGW firmware 0.61 gave an error.

rvdbreemen, 2021-01-21:
@STemplar would you consider joining us on Gitter.IN. https://gitter.im/OTGW-firmware/community

STemplar, 2021-04-06:
I checked all Data-ID's and missed some bit descriptions:

```
ID0:HB5: Master status: Summer/winter mode
ID0:HB6: Master status: DHW blocking

ID0:LB7: Slave Status: Electricity production

ID2:HB0: Master configuration: Smart power

ID3:HB6: Slave configuration: Remote water filling function
ID3:HB7: Heat/cool mode control

ID4 (HB=1): Remote Request Boiler Lockout-reset
ID4 (HB=2): Remote Request Water filling
ID4 (HB=10): Remote Request Service request reset

ID70:HB0: Master status ventilation / heat-recovery: Ventilation enable
ID70:HB1: Master status ventilation / heat-recovery: Bypass postion
ID70:HB2: Master status ventilation / heat-recovery: Bypass mode
ID70:HB3: Master status ventilation / heat-recovery: Free ventilation mode
ID70:LB0: Slave status ventilation / heat-recovery: Fault indication
ID70:LB1: Slave status ventilation / heat-recovery: Ventilation mode
ID70:LB2: Slave status ventilation / heat-recovery: Bypass status
ID70:LB3: Slave status ventilation / heat-recovery: Bypass automatic status
ID70:LB4: Slave status ventilation / heat-recovery: Free ventilation status
ID70:LB6: Slave status ventilation / heat-recovery: Diagnostic indication

ID74:HB0: Slave Configuration ventilation / heat-recovery: System type
ID74:HB1: Slave Configuration ventilation / heat-recovery: Bypass
ID74:HB2: Slave Configuration ventilation / heat-recovery: Speed control

ID86:HB0: Remote ventilation / heat-recovery parameter transfer-enable: Nominal ventilation value
ID86:LB0: Remote ventilation / heat-recovery parameter read/write : Nominal ventilation value

ID101:HB012: Master Solar Storage: Solar mode
ID101:LB0: Slave Solar Storage: Fault indication
ID101:LB123: Slave Solar Storage: Solar mode status
ID101:LB45: Slave Solar Storage: Solar status

ID103:HB0: Slave Configuration Solar Storage: System type
```

rvdbreemen, 2021-05-01:
@STemplar will be fixed in next release 0.8.4. Thanks for checking all codes for me ;-) Checkout dev-ventilation for all msgids that where missing. Still need to redo the home assistant integration, but it's all there now as far as I can see.

rvdbreemen, 2021-05-01:
here: https://github.com/rvdbreemen/OTGW-firmware/tree/dev-ventilation

rvdbreemen, 2021-05-02:
@STemplar I think I fixed all the missing bits and pieces now. So if you would be so kind to review my work, and confirm or deny it. I will close it, if you find anything, then we will reopen this issue.

closed 2021-05-02 by rvdbreemen