rvdbreemen/OTGW-firmware #6: add encrypted MQTT option

vampywiz17, 2021-01-11:
Hello there,

Now is possible to set the MQTT server via UI and that is wonderful! But now, it only support a unencrypted connection. So if we would like to a secure connection a remote GW, need a relay server locally. (it will be a problem, if that is a weekend house and no any infrastructure, only internet connection)

So please, if possible, add a SSL option to you firmware.

Thanks!

Some examle:

https://github.com/flespi-software/ESP8266_mqtts

https://gist.github.com/eLement87/133cddc5bd0472daf5cb35a20bfd811e

rvdbreemen, 2021-01-11:
Need to check out: https://arduino.stackexchange.com/questions/72684/how-to-connect-to-mqtt-broker-with-tls

vampywiz17, 2021-01-12:
I read you link, seems to the encrypted but not signed method is easier. Possible that is enough with first step. It good for own MQTT broker.

rvdbreemen, 2021-01-12:
@vampywiz17 that's what I thought when I was reading the linked page. So, I think we will go for that, it should do in most situations if it works. It's the ONE feature you want and then the project is done ;-)

DaniilMir, 2022-11-23:
A nice to have feature for secure environments or for the ones who are using cloud MQTT servers.