rvdbreemen/OTGW-firmware #7: Support of DS18B20 sensoren connected to nodeMCU

mlommers, 2021-01-12:
It would be great to also support DS18B20 sensoren directly connected to the nodeMCU

-configure GPIO pin with connected OneWire sensors
-read out temp value (per connected sensor)
-sent (each) temp value over mqtt

With the default configuration of the nodo shop configuration, ESPEasy is supported.  However ESPEasy doesn't allow other devices connected to the nodeMCU when it is configured to readout the OTGW connected PIC via serial interface. As next to the readings of the boiler via OTGW I also want to readout the temperatures of each underfloor heating group I currently need to place an extra ESP device only for this purpose. When possible I would like to combine both ESP purposes to one ESP configuration

Roos-AID, 2021-01-12:
@mlommers I can confirm that the original ESPEasy https://github.com/letscontrolit/ESPEasy does support OneWire sensors. This is my current setup, I even have a pulse counter in my setup (which only works until version August 2019 due to bug in ESPEasy after that. 
See : 

[image]

In order to make it work, I have hardwired the connections to the ESP board. 

However, I do support the new development of Robert as this gives us much more flexibility in the monitoring as we directly can feed MQTT with his solution.  

It is a good suggested enhancement to add OneWire to the new OTGW-firmware, but I do have concerns on  memory and processing power of the ESP8266 in this case.

mlommers, 2021-01-12:
@Roos-AID: strange i couldn't get OTGW up and running with ofther devices configured (or the other way around no temp reading when OTGW configure), after that I also read the comment in the installation manual of nodoshop (dutch) un heading Devices stating "When used for OPENTherm, only 1 device (OPENTherm) may be added here, so leave the positions empty"
Which Version are you using? I'm using the recommended version: https://www.letscontrolit.com/esp147.zip

nevertheless would be great if this initiative would support DS18B20 sensors. I like the approach rvdbreemen is taking

rvdbreemen, 2021-01-12:
@Roos-AID Could you explain to me how you setup the hardwiring. If that's possible with the OTGW hardware, then I can support it. Software-wise it's a very limited memory use, as it is mainly the two-wire library that we need to support DS18B20 sensors.

If you share you hardware fix then I can estimate if it's worth supporting it in my firmware.

vampywiz17, 2021-01-12:
Other idea (but similar with this idea)

External temperature sensor addon (like a DS18B20), not connect boiler but to nodeMCU. it automatically send the external temperature data to a specific OT ID (because it possible to send temperature value via OTGW. HA and other implementation is also do this)

Roos-AID, 2021-01-12:
@vampywiz17 I would not recommend to have multiple DS18B20 connected and sending values to OT IDs.  This is only supported for a single outside temperature probe.  All other OT IDs might have a strange effect on the boiler or other connected devices on OT.  I the latest OT V3 there are a couple of OT IDs that are non standard and might be used for other purposes, but you can't predict how the boiler and other devices might react to it. 

@rvdbreemen Robert, I have used this schematic to connect the DS18B20, but used D7 (GPIO 13) iso D5 as in the schematic.

[image]

Roos-AID, 2021-01-12:
@mlommers The NodeMCU version I use is : mega-20190827 .   This is working for over a year now. The only problem is that sometimes the Wifi connection is lost, but a simple restart solves this.  This might be fixed i a newer version of the NodeMCU. 

In order to update the NodeMCU to this version, you must detach it from the OTGW and flash it with full erase first. The you have to make sure that all parameters, incl. the watchdog are set correctly as per original instruction.

vampywiz17, 2021-01-12:
@Roos-AID 

I only want to connect one DS18B20 and send it to a correct ID, It affect any problem?

rvdbreemen, 2021-03-14:
Added in release 0.8.1 by @sjorsjuhmaniac

rvdbreemen, 2021-03-14:
@mlommers issue closed

mlommers, 2021-04-10:
just got a new nodeMCU to try this firmware on. Installed OTGW-firmware and connected a DS18B20 sensor. Configured MQTT and checked "GPIO Sensors Enabled" to true
However I don't see the vallues passing by over MQTT.
I didn't connect the nodeMCU on the OTGW board, but i won't expect that that is needed. Correct?
I also noticed that after reboot the "GPIO Sensors Enabled" is no longer checked.
Any idea what is wrong and or what else I need to do to see the DS18B20 sensor values being sent?

rvdbreemen, 2021-04-10:
Hmm, will have to dig in and find out what is going on. But the detection of the sensor is only during reboot, but you say the “enable” is turned off after reboot. 

So not sure what is going on here.

mlommers, 2021-04-10:
Could it be that the enable is turned off when no sensors are detected?
Now it is working, the sensor is found on startup, en the enabled stays on after reboot. however the interval time is not pertained after reboot and fallsback to 5 seconds

rvdbreemen, 2021-04-10:
Not sure what’s going on. And not near my laptop this weekend so now way of checking. But glad it’s working now for you. 

When I get back, I’ll check out what is going on with the interval setting. For now 5 second will have to do. I’ll fix that once I get to it. I’ll reopen this issue just to make sure.

sjorsjuhmaniac, 2021-04-10:
> just got a new nodeMCU to try this firmware on. Installed OTGW-firmware and connected a DS18B20 sensor. Configured MQTT and checked "GPIO Sensors Enabled" to true
> However I don't see the vallues [...]

Correct, is not needed.

> I also noticed that after reboot the "GPIO Sensors Enabled" is no longer checked.

This is not by design but could happen if there are no sensors detected on boot AND something triggers saving the config (because all current VAR states will be written to the config file when that is called). This is the error you should see if there are no sensors detected and this feature is disabled until the next reboot (or not if something or someone triggers a 'save config'), see:
https://github.com/rvdbreemen/OTGW-firmware/blob/8679dbb7c83545c64c082c9f59a0287ccd62b95c/sensors_ext.ino#L74-L79
This should only set the var to false but not change the setting. We'll need to dig into this to check where it goes wrong.

> @rvdbreemen or @sjorsjuhmaniac: Any idea what is wrong and or what else I need to do to see the DS18B20 sensor values being sent?

based on the above i suspect a bad connection or sensor layout if you use multiple. There's no need to do anything besides enable it in the config, set the correct PIN, connect the sensor to the correct PIN and reboot.

> 
> on the telnet session i see:
> [16:00:35][ 14712| 12464] pollSensors ( 93): No Sensors Found, please reboot the node to search for sensors

This is coming from inside the polling loop. 
https://github.com/rvdbreemen/OTGW-firmware/blob/8679dbb7c83545c64c082c9f59a0287ccd62b95c/sensors_ext.ino#L91-L95

Technically it could mean one of two things
- a sensor was found during boot (and the sensor var is still set to true) but is 'no longer available'.
- this msg was printed after you enabled the setting but didn't reboot yet.

I think 1) is not possible because `numberOfDevices` only gets sets once in the code, when `initSensors()` is called on boot.
https://github.com/rvdbreemen/OTGW-firmware/blob/8679dbb7c83545c64c082c9f59a0287ccd62b95c/sensors_ext.ino#L53

> 
> but after rebooting the "GPIO Sensors Enabled" is no longer checked
> enable that one I get no sensors found again
Depending on WHICH 'no sensors found' msg you get this is expected behavior. Directly after enabling the option `pollSensors()` is called (it's always called but now executed because the setting var is set to true) and 'fails' because var `numberOfDevices` is < 1. As mentioned above, `numberOfDevices` only gets set on boot.

> 
> Just notice that in the settings.ini file the following line is kept after reboot: "GPIOSENSORSenabled": true,
> however this is not the state in the settings settings UI. The GPIO Publish Interval (sec) is also no retained after reboot in the settings UI, I don't see this setting in the settings.ini file.

We'll need to look at this.

> 
> nervertheless I still don't see the DS18B20 sensor values
> 
> I finally got it to work however the GPIO Publish Interval (sec) is not retained after reboot

Right now, without looking in detail at the code:
- looks llike you might have a bad sensor, bad sensor connection or bad network layout
- i set the config var to false to prevent excissive failure logging when no sensors are found. I can happen that the config_safe void is called after this and thus the setting now gets saved with state 'false'. This is a bug.
- there is an issue with the interval setting being saved (or not)

sjorsjuhmaniac, 2021-04-10:
@mlommers pls share the full log from a reboot where this behavior happens.

mlommers, 2021-04-11:
@sjorsjuhmaniac: thanks for your extensive explenation. I think you are right, at the start somehow the sensor was not seen turning the enabled into false which is shown in the settings UI aswell. In the settings.ini the variable was still true until I or something else triggered the save settings.
In the end I got the sensor working ans I didn't see the problem with the enable variable anymore.

The other issue with the interval is most probably that this setting isn't saved and made persistent in the settings.ini file

rvdbreemen, 2021-04-11:
And @Roos-AID just did a PR to fix that for the next release.

rvdbreemen, 2021-05-02:
Fixed in release v0.8.4

closed 2021-05-02 by rvdbreemen