rvdbreemen/OTGW-firmware #8: Javascript error

DaveDavenport, 2021-01-12:
I have updated to latest git (self compiled) and tried latest release v0.6.1 and i get the follwing javascript error:

on firefox:
```
Uncaught TypeError: document.getElementById(...) is null
    bootsTrapMain http://192.168.1.151/index.js:37
    EventHandlerNonNull* http://192.168.1.151/index.html:61
index.js:37:14
    bootsTrapMain http://192.168.1.151/index.js:37
    (Async: EventHandlerNonNull)
    <anonymous> http://192.168.1.151/index.html:61
```
on chrome:
```index.js:37 Uncaught TypeError: Cannot read property 'addEventListener' of null
    at bootsTrapMain (index.js:37)
bootsTrapMain @ index.js:37
load (async)
(anonymous) @ (index):61
```
the page stays empty for the rest.
I checked that the version of the index.js is 0.6.1.

I re-uploaded the spiff.bin

DaveDavenport, 2021-01-12:
reflashed the whole nodemcu and it seems ok now.

closed 2021-01-12 by DaveDavenport

rvdbreemen, 2021-01-13:
@DaveDavenport glad it was fixed.