rvdbreemen/OTGW-firmware #10: Human readable ASF, RBP-flags, TSP, FHB-values, OEM fault/service codes and the like.

STemplar, 2021-01-19:
The Remeha boilers (and other brands too) can send fault and parameter codes to the thermostat.  It is done by using the Transparent Slave Parameters. Only Thermostats of the same brand know which fault or parameter codes corresponds with a certain OEM fault code / (TSP) value.

A nice feature woult be being able to make a JSON-file with a crosstab which pairs the OpenTherm values and the corresponding description from the manuals.

# Class 1: Control and Status Information
register 5	        {flag8 u8}	"Application-specific flags" (correponds with the Remeha E:xx fault codes.)
register 72	{flag8 u8}	"Fault flags/code V/H"
register 73	{u16}		"OEM diagnostic code V/H"
register 102	{flag8 u8}	"Solar storage fault flags"
register 115	{u16}		"OEM diagnostic code"

# Class 2: Configuration Information
register 3	{flag8 u8}	"Slave configuration"
register 74	{flag8 u8}	"Configuration/memberid V/H"
register 103	{flag8 u8}	"Solar storage config/memberid"

# Class 6: Transparent Slave Parameters
register 11	{u8 u8}		"TSP setting" (correponds with the Remeha Pxx parameter codes.)
register 89	{u8 u8}		"TSP setting V/H"
register 106	{u8 u8}		"TSP setting solar storage"

# Class 7: Fault History Data
register 13	{u8 u8}		"Fault buffer entry"
register 91	{u8 u8}		"Fault buffer entry V/H"
register 108	{u8 u8}		"Fault buffer entry solar storage"

rvdbreemen, 2021-01-19:
@STemplar I use a lookup table that does that. I actually used the spec 2.2 for the description of the ID. Is that what you mean? Because I am planning to include all the ID's you provided in the other ticket.

STemplar, 2021-01-20:
> 
> 
> @STemplar I use a lookup table that does that. I actually used the spec 2.2 for the description of the ID. Is that what you mean? Because I am planning to include all the ID's you provided in the other ticket.

No, I mean these codes: 
[image]
[image]
[image]
[image]

You cab find them in the Remeha installation manuals ands appear to be generic for Remeha boilers. Remeha Thermostats can link those descriptions to the corresponding Data-ID replies.

Other Remeha info:
Data-ID 5: correponds with the Remeha E:xx fault codes.
Data-ID 11: correponds with the Remeha Pxx parameter codes.
Data-ID 35: reported value is fan speed in rpm/60 .
Data-ID 115: correponds with the Remeha Status and Sub-status numbers using {u8 u8} data-type.

Checking this [Equipment matrix](http://otgw.tclcode.com/matrix.cgi) I suspect Itho Daalderop, Intergas and Atag to have their own implementations.

rvdbreemen, 2021-01-20:
Aha, so I have not looked at the implementation of the The Slave Parameters. I was focused on the basic message first. But it's definitly something I can consider implementing, as I own a Remeha Calenta myself ;-) I will reopen the issue, and put it back on the backlog.

STemplar, 2021-01-21:
> 
> 
> Aha, so I have not looked at the implementation of the The Slave Parameters. I was focused on the basic message first. But it's definitly something I can consider implementing, as I own a Remeha Calenta myself ;-) I will reopen the issue, and put it back on the backlog.

Then you probably have an iSense thermostat too. Check every parameter and option while checking the send and returned Data-ID's. Then you can compile a crosstab.

STemplar, 2022-01-09:
Hi Robert, here my first attempt:
[OTGW Remeha lookuptable.txt](https://github.com/rvdbreemen/OTGW-firmware/files/7835905/OTGW.Remeha.lookuptable.txt)

STemplar, 2022-01-09:
Master MemberID Code (Data-ID 2 LB):
Remeha <11>
Master product type and version of twee Remeha thermostats (Data-ID 126 u8 u8):
qSense v1.2 <35 12>
iSense v27 <32 27>

rvdbreemen, 2022-01-09:
So we need a way to collect information ;-)

STemplar, 2022-01-09:
Indeed, perhaps the easy way out is display the byte fields andd ask for people to post the accompanying thermostat model info in the Discord channel?