rvdbreemen/OTGW-firmware #13: Blynk integration - just a thought

suenio, 2021-01-29:
Robert first of all great work on this software. Did you thought about expanding it with integration to Blynk ?

rvdbreemen, 2021-01-29:
@suenio thank you so much for the compliment. What would I need to do to integrate with Blynk? I do not use that personally, could you give me pointers to relevant information, so I can investigate?

suenio, 2021-01-29:
Best to start with blynk is to look at https://blynk.io/en/getting-started. Integration is quite simple however concernig maybe how much it will impact stability of integration with OTGW due to integration library implementation.

I was thinking about ability to expose status and temperature information such way as it will also give ability for non Home Automation users to control their thermostats remotely as well as command API so you could modify for example room setpoint through app.

rvdbreemen, 2021-01-29:
I did a quick scan of blynk. I get what they do, I personally do not like working with cloud-based solutions, simply because I want all data to be mine. But I do get where you're thoughts go this way. It would enable non-home automation users to use the OTGW easily. 

Let me think about it. Maybe that would be a separate project all together.