rvdbreemen/OTGW-firmware #14: Lost information after wemos restart

suenio, 2021-01-30:
From soem reasons my wemos reatarts 2-3 times a day. There is no significant impact due to that an of course it is no surrprise that after restarting wemos it will lose data capured from OTGW but is it possible to have configurable routine that will execute set of predefined opentherm read commands so rather than wait for thermostat to do so values will be available after wemos restart. I can achive same setting soem commands in HA to be sent to Reat API however it woudl be nice to have such configurability and self "recovery" after restart.

I know other option could be to store latest values in memory however not sure it is best approach due to flash lifecycle in such case.

rvdbreemen, 2021-01-30:
@suenio not sure why your hardware resets. The fact that after every restart the data is gone, is because the data is gone from memory. In the nodoshop version, every reset also resets the PIC. So all data in both the NodeMCU and the PIC are gone. You cannot use the commands to fetch data.

The only way to recover after a reset like you suggest, is indeed to store it in flash. But that would cause an issue with flashlifecycle. So I am not going that way.

However how bad is the situation for you, normally within about 1 minute maximum all values should be recovered by itself. Is there anything that "goed wrong" because of this. One solution could be to use MQTT, as the hub is not reset and persist the state too.

suenio, 2021-01-30:
Usually it restarts evey 2-3 hours. You are right that most values are back within couple of minutes however for example if my thermostat announces and changes room temperature setopint only 3 times a day there is couple of hours between this value is beeing visible.

rvdbreemen, 2021-01-31:
Hmmm, well the data could be persisted in flash. The problem that I have with that is that we would start wearing down the flash, for the persistence of some data. Do you have any idea why it resets? Could you do some logging (on port 23) and show me the reset moment. You could use putty to log. Also what is reset reason on the device information page? And please let me know the reset count in the file (reboot_count.txt). You can use the File explorer to just look at the content of that specific file.

Thanks for letting me know more,
Robert

suenio, 2021-01-31:
First of all my value in reboot_count.txt is 174 and going up after every restart. I think I know what is going on. Looks like when MQTT is setup but with wrong credentials (in my case problem is too short password field size) NodeMCU hangs and restarts. So this is most likely false alram but maybe good would be to handle this as connection error with info about wrong credentials rather than triggering NodeMCU restart

suenio, 2021-01-31:
Here are port 23 logs and last line when restart happens:
[00:48:00][  23656| 23376] processOTGW ( 642): msg=[C: 00:48/1] value=[0000000c]        type[  0] id[  0] hb[  0] lb[ 12]       [READ_DATA       ]      [Status                        ]
[00:48:00][  24008| 23592] handleMQTT  (  93): MQTT State: MQTT try to connect
[00:48:00][  23336| 22944] handleMQTT  (  96): Attempting MQTT connection .. Username [homeassistant]

rvdbreemen, 2021-02-01:
Okay, it sounds like a check that needs to be added, I agree failing to authenticate should not result in a reboot.

So I will mark is as a bug for now, and try to fix it in a future release.

rvdbreemen, 2021-03-01:
Changing the reconnect to MQTT to 4 seconds, so the watchdog does not reset the device. This should prevent the resetting when disconnected. Thanks to @sjorsjuhmaniac point out the obvisous. And to @tjfsteele for finding the same thing in another testing frenzy.

closed 2021-03-01 by rvdbreemen