rvdbreemen/OTGW-firmware #15: No web UI after clean flash 0.7.0

Nauttor, 2021-02-02:
Hi, 
I flash both files on the 0.7.0 release on my nodemcu, it flash it as expected with no error. Afterwards I see the OTGW node active and I set the wifi successfully. Then I tried to connect to the web UI via IP address to setup the Mqtt but it does not response, like the web server is not active. Am I doing something wrong?

Thanks

Nauttor, 2021-02-02:
OK, After a few hours looking around it, I discover that actually it's a bug. 

The problem is that I have all my IoT devices in a segregated Vlan without access to the internet and firewalled. Because of that your firmware try to reach the NTP server, and if it does not reach the server it does not continue loading the web UI and every else it stay in a loop trying to connect to the ntp server. 

Please fix that when you possible, Thanks.

Nauttor, 2021-02-02:
Hi, 

I see that I can't either set the timezone by hand. 

About the Mqtt client I see that it publish all the data like:
homeassistant/binary_sensor/OTGW/........

I know that's for the home assistance autodiscivery etc, but not all the people use HA, I use Openhab, It would be great too to have an option to personalise the MQTT topic, and a check for example to use or not HA automation.

Thanks

rvdbreemen, 2021-02-07:
The firmware behaves as needed. 

Two feature requests:
1. Optional NTP synchronisation. For planned InfluxDB it is a requirement, but I guess we can disable features if dependencies are not matched.
2. Port to ESP32. I could do that ofcourse, then you have to design your own hardware. Do you have a ESP32 version of the OTGW gateway at this point? I have ported another project over to ESP32, so it's doable.

Nauttor, 2021-02-07:
> The firmware behaves as needed.
>
> Two feature requests:
>
> 1. Optional NTP synchronisation. For planned InfluxDB it is a requirement, but I guess we can disable features if dependencies are not [...]

Hi @rvdbreemen ,

Thanks for your reply. Let me disagree on the NTP, it does behave as you want to behave, not as it should. The ntp as you said can be a requisite for influxdb to work, but not all the people want to use that feature, the "main" feature of a OTGW wifi node firmware should be give access to the Ser2net or mqtt access to the OTGW. So in order to work the OTGW wifi firmware does not need to access NTP, and in this case if it does not connect, you are hang in a loop indefinitely. At least, as you said it would be great to have the option to configure the NTP manually AFTER the node finished the boot sequence and you can access to the web page. So, optional NTP should be great. 

About ESP32, I have 2 esp32 modules, one ESP32-WROOM-DEVKIT and one ESP32-Cam which is a great module because it have an external antenna connector which enables you to extend the range. In my case my boiler is far away from my AP and server and I can't install an ethernet cable, so I need to use an external antenna to make it stable. At the moment I connect the ESP32 with tiny dupont cables to the NodoShop board for testing, and it works with ESPEasy firmware on the esp32. It would be great to have a compilation for that board too as Is a more capable hardware. Thanks.

rvdbreemen, 2021-02-11:
You have a good point. I will add a NTP enable/disable feature in a future release.

Nauttor, 2021-02-11:
> You have a good point. I will add a NTP enable/disable feature in a future release.

[image]

Hahahaha, thanks for your consideration :)

rvdbreemen, 2021-02-14:
Feature will be in 0.7.6, currently in dev. Added another much wanted feature, enablement of setting up the TZ itself.

closed 2021-02-14 by rvdbreemen

Nauttor, 2021-02-14:
@rvdbreemen , the ESP32 port milestone appears as 100% complete it's an error or the 0.7.5 will be esp32 compatible?

rvdbreemen, 2021-02-14:
That's a mistake. As I closed this ticket because of TZ not of ESP. I will try and correct it.