rvdbreemen/OTGW-firmware #16: SendCommand with MQTT

Nauttor, 2021-02-02:
Hi, 

I tried to send OTGW/command/CS=52 but nothing happens on the OTGW side, can you explain how can we send commands via MQTT or API. Thanks

ikkemaniac, 2021-02-06:
The method you used to send commands is correct. There is no guarantee that the OTGW has successfully received and processed the command due to way the serial communication is setup. Unless the OTGW prodes a reponse on the serial bus, it has not processed it successfully.

Note that the OTGW itself (so NOT this firmware package) should respond to a command sent by replying the same command with a ':' so for 
```CS=52```

you should see the OTGW respond with 
```CS: 52``` 

Do you see the OTGW responding?

rvdbreemen, 2021-02-07:
@Nauttor indeed the response to a command it is not checked at this point. So the OTGW could have responded with an error for example. So the current release just sends command, and you have to check yourself. I will add this feature in a future release, so you can check what the response was to command.

Nauttor, 2021-02-07:
> @Nauttor indeed the response to a command it is not checked at this point. So the OTGW could have responded with an error for example. So the current release just sends command, and you have to [...]

Hi @rvdbreemen and @ikkemaniac,

I see in the telnet debug cli that the command is send, but as I don't see any response and my boiler don't change the state I image the message does not reach the boiler or something happens in between.  As nothing happens when I sent the command via MQTT I tried it connecting the firmware to the OTMONITOR, and I tried again to send the commant from there. I see in the log OTmonitor published CS:52 but, again nothing happens. As I said is like the message was never sent or published in the TX of the board. 

Please @rvdbreemen , can you published and updated Binary of the las 0.7.3? thanks.

ikkemaniac, 2021-02-07:
@Nauttor It's not uncommon that commands have be to send multiple times before they are picked up by the OTGW, this is due to the way the serial communication works.

Now when you DO get a ```CS:52``` response, you should see something happening on the boiler end if conditions in the boiler are right. 
My boiler, when it's normal stand-by state, responds within 10 seconds to an acceptable CS command. Note that not every CS value is 'acceptable' and if it's below certain thresholds, or within x amount of time after a previous burn or DWW tap, the boiler won't respond due it's own, internal, software.
To test I would advise you to send a CS=80. This to ensure it's sufficiently high to be above any other threshold. Note this is also the default behavior of an iSense thermostat.
Another thing to specifically mention is the boiler status. My boiler provides a ton of status codes on it's display, that aren't available via OT, to show it's internal state (for example 'initiating burn' or 'cooling down'). These states can also help you identify why the boiler might not respond when you send a CS.

The best time to test if the command actually works for you/your boiler is to issue it when the boiler is already in a Central Heating  Burn for a couple of minutes. You should see your thermostat send out a fluent CS curve. Take the active setpoint (anything below 60), add 5 and sent the command, you should see a response within a minute.

last note: all of the above is beyond the scope of this firmware.

Nauttor, 2021-02-07:
Hi @ikkemaniac, 

Let me reformulate the cuestion as I think I didn't explain it clearly. 

My otgw nodoshop hardware works with the Nodemcu and the EspEasy Ser2Net service configured. I can send and receive the commands from the boiler without any issue. However that's not the case with OTGW-firmware 0.7.0. I can read the data sent from the boiler but my commands never reach the boiler. 

BTW, the project name is quite confuse, when you read on the forums about it you don't know always if we are speaking about the OTGW HEX firmware or the ESP8266 one, a re-consideration on the name may be help.

rvdbreemen, 2021-02-07:
@Nauttor I understand your confusion. I only realized that it might be confusing much later. As the PIC firmware and the ESP 8266 are both firmware.

I am working on an update, that includes a reliable firmware upgrade feature. That's the reason why it's taking a bit longer.

Could you capture some debug log and post it here, so we can see what is going on. Just to make sure I understand, you use serial2net port to communicate (not MQTT or REST). If so, just know that the serial communication is a simple proxy, forward al bytes to the device and back.

Hope to hear from you,
Robert

Nauttor, 2021-02-07:
@rvdbreemen, When I use the Ser2Net service with the ESPEasy it works, sort of "stable" because as I said my node is far from the AP and sometimes is not 100% stable. I can send and receive without any problem. If I tried with your firmware I can read(receive) from the OTGW but every time sent(transmit) nothing happens on the boiler. I tried several codes like MM=50, CS= 65, etc.  I didn't get any error, it's like even if theoretically is sent, and the log show that, the value is not sent or is not received by the boiler. Te result is the same on mqtt, api, or ser2net via OTMonitor. 

About the logs, sure, I'll collect some of them.

rvdbreemen, 2021-02-15:
Hmmmmm @Nauttor that's strange. I will have to investigate myself a little more. Sorry about the delay, been working on the update of the pic firmware, before I solve other issues.

Nauttor, 2021-02-15:
Don't worry I'll try to replicate it with 0.7.5.

rvdbreemen, 2021-02-28:
@Nauttor has your problem with sending commands been fixed?

rvdbreemen, 2021-03-25:
@Nauttor think you will like release 11th v0.8.2 it contains command queue for all msg via MQTT. Please validate the function for me.

closed 2021-03-25 by rvdbreemen

Nauttor, 2021-03-25:
> @Nauttor think you will like release 11th v0.8.2 it contains command queue for all msg via MQTT. Please validate the function for me.

Hi @rvdbreemen , I've been working offshore the last month, so it's difficult for me at the moment to test it, I'll hope to be at home soon. Thanks for your effort, I'll test it as soon as I return. Best regards