rvdbreemen/OTGW-firmware #22: Feature Request - support relay module

vampywiz17, 2021-02-11:
Hello there,

On NodeMCU, a lot of free GPIO is exist if we connect OTGW. I thinking that possible to integrate a relay module, that working pararel with CH state. This state basically is state of the boiler internal pump ON/OFF.

Or if we use two relay, it possible the do the same with H2 parameter.

The Reason is that lot of boiler not support official relay module (some yes, like Baxi) and in this case the people use pipe thermostat, to turn on/off floor heating external pump, or heating circle valve.

sjorsjuhmaniac, 2021-03-20:
@vampywiz17 , what is the H2 parameter? Did you mean CH2?

vampywiz17, 2021-03-22:
@sjorsjuhmaniac 

Yes, CH2 (second heating cycle)

rvdbreemen, 2021-04-03:
@sjorsjuhmaniac added the gpio ot hook feature, it will be in releas v0.8.3

closed 2021-04-03 by rvdbreemen