rvdbreemen/OTGW-firmware #26: Time is hour behind

PvdGulik, 2021-02-20:
Hello,

Is seems that the OTGW-firmware is a hour behind. Due to DST-setting?
I don't know where is get its time from, but all my other ESP's get the right time.
And the boiler time is also correct.

Thank you!

rvdbreemen, 2021-02-21:
Check your settings page, there you can setup the correct timezone. It does DST correctly.

rvdbreemen, 2021-02-21:
I reopened, to wait for confirmation that it works correctly. @PvdGulik

PvdGulik, 2021-02-21:
Sorry, but where in the settings page?
There's no option... See picture.[cid:F6983424-0F20-4DEA-9736-292D7E52AA92]

Thank you!

vriendelijke groet,

Pieter

rvdbreemen, 2021-02-21:
I think I know what's going on. Sorry about that. In the dev release there is a setting.
And I just found that sometimes there is DST and sometimes it's not.
It's under investigation.

Thanks for reporting this @PvdGulik

PvdGulik, 2021-02-22:
Solved in v0.7.6, thanks!

rvdbreemen, 2021-02-22:
fixed in v.0.7.6 and beyond

PvdGulik, 2021-02-24:
Hello,

It seems that problem is not totally resolved.
This morning (ca. 9:30) after updating to lateste version 7.7,
the NTP timezone was CET,and time was correct.
Now, at 20:30 the timezone has changed by itself to Europe/Amsterdam and is one hour behind again....

rvdbreemen, 2021-02-24:
Hi @PvdGulik thanks for reporting back. I am hunting that bug, not sure why it is happening. The simpel fact that it changes from CET to Europe/Amsterdam is telling me that it is switching to the default setting, which is now Europe/Amsterdam. 

The one hour behind is really weird, as my timezone is setup exactly the same and reporting the correct time.

Hmmmm, what is going on.

rvdbreemen, 2021-02-24:
@PvdGulik 

Well, I changed around some stuff in the dev branch. Now it will not try to be smart anymore and detect the timezone. I will now just accept whatever you decide the Timezone is.

Could you please verify this version? Let me know if you need a binary to test it.
Thanks
Robert

PvdGulik, 2021-02-24:
Yes, no problem to test. Where I can find the new version of the ino.bin?

rvdbreemen, 2021-02-24:
Here it is:
[OTGW-firmware.ino.bin.zip](https://github.com/rvdbreemen/OTGW-firmware/files/6039111/OTGW-firmware.ino.bin.zip)

PvdGulik, 2021-02-24:
Hello Robert,

I installed the new bin, set Timezone to CET, and saved.
Time was 1 hour behind. Reboot.
After reboot Timezone is set back to Europe/Amsterdam, but time is correct.
I set timezone back to CET, save but won't reboot.

Report you the results tomorrow.

Pieter

PvdGulik, 2021-02-25:
Hello Robert,

8:51 Time is still correct, but timezone is back to Europe/Amsterdam.

I'll wait a couple of hours, keep you informed of any changes.

rvdbreemen, 2021-02-25:
Sounds like it is fixed so far ;-)

closed 2021-02-25 by rvdbreemen