rvdbreemen/OTGW-firmware #30: MQTT publishing intervals

ReneHonig, 2021-03-08:
@rvdbreemen et. al., thanks a lot for efforts on this nice project, much appreciated!

In the current setup values are published to the MQTT server at a pretty high frequency (multiple times/sec). This is nice for debugging, but puts quite some load on the system and is probably not really necessary for most users.
Would it be an idea to set a publishing interval on the data (e.g. every 30 secs) and maybe have a debugging mode for publishing values like otmessage?

Unrelated, but maybe of interest: there is an excellent and very [mature projec](https://github.com/proddy/EMS-ESP)t that has comparable functionality for the Bosch (and related brands like Nefit) EMS bus. It might be beneficial to talk to @proddy, the creator of that one to compare notes.

rvdbreemen, 2021-03-08:
@ReneHonig the MQTT topics are generated based on the incoming OT message, they are publish as they come in. Some msg translate in multiple updates (like the status message). The frequency of those message I do not control at this point. I understand what you are asking for, a "rate limited" publishing. However this would change the 'realtime' nature of the current design. A 'flame' status could be there for less than the interval you set, or, stay a lot longer than that. 

The publishing of the otmessage I could make optional, I agree on that btw. The reason I added it, is simply because the original OTmonitor app (by Bron Schelte) had that same function. I wondered if anyone would even do something with it, it is very usefull for recording the incoming OT msg stream however. Making that optional would be something I would consider.

Recap:
- interval setting -> maybe in the future
- debugmode -> can do 

For the related project, thanks for the link. Will look at it.

rvdbreemen, 2021-03-09:
I played around with pushing on change model. Thanks to Sjors for putting it in my head to try. But I am running into a wall. The publishing of topics only on change will not work in the end, as you could have setpoints that would only be published once. 

For now I don't see the value of intervals or the push-on-change only, as the sideeffects are not things you want.

If you don't want to topics, then don't subscribe to them. If you need sensors not to appear in Homeassistant you can simple delete the sensors from the mqqtha.cfg file. Just the delete the sensors you do not need. That simple.

The otmsg setting is added to the current dev branch, and will come along with the next relese.

rvdbreemen, 2021-03-10:
@ReneHonig thanks for the links. Looks interesting indeed, it's definitly something to investigate. Also it seems to the project is very simlar to what I am trying to do, it's for the EMS bus, so I have to investigate.

I can look at what he did, and then reconsider the solution I take.

bluewalk, 2022-02-14:
@rvdbreemen I am seeing loads of publishes to topics like `otgw/value/otgw-XXX/ch_enable`, `otgw/value/otgw-XXX/centralheating` etc with the same value over and over.

Would it be possible to add a little if statement comparing the new value with the old value and __only__ publish when the value changes?

Reason I'm asking is that software using these topics get flooded with the same value multiple times a second.

rvdbreemen, 2022-02-14:
@bluewalk wat software are you using? The topic seem to come from the OTmonitor software and not from the ESP firmware. I am correct?

rvdbreemen, 2022-02-14:
Rereading your request. Your suggestion has been asked before. The topic is frequently updated because the status is send frequently. What is the reason this might be a problem for you?

bluewalk, 2022-02-14:
> Rereading your request. Your suggestion has been asked before. The topic is frequently updated because the status is send frequently. What is the reason this might be a problem for you?

At the moment it's flooding logs of my domotics and updating the value in my domotics with the same value seems unnecessary and redundant (beetje onnodig). Aside from flooding the MQTT server but it's supposed to be able to handle that though.

bluewalk, 2022-02-14:
Great, thanks! If I have time I might have a look and create a PR 👍

rvdbreemen, 2022-02-14:
If you have some time to spent then a PR is more than welcome. 

Actually there is a data model for all values of OpenTherm. I even have an old branch for publish on change logic. I really need to refactor that code.

bluewalk, 2022-02-15:
I did a quick hack of adding a check to each `is_value_valid` in the print functions to compare `value` to either `_value` or `OTdata.u16()` which seems to work. It ain't pretty but functions for now.

Looking forward to your refactored code regarding sending the MQTT messages :-)

HenkRenting, 2025-03-27:
Any news on this?
The updates are flooding my MQTT/Homey as well...
Would be great if you could set an interval or filter which messages are published and which not.