rvdbreemen/OTGW-firmware #32: MQTT broke?

DaveDavenport, 2021-03-14:
I flashed latest version (0.8.1).
I decided to do a clean update, so erased the flash, flashed both files of the 0.8.1 using the command from the wiki.

Logged into wifi, configured wifi.
Rebooted and logged into web and configured mqtt and the others.
I can see the webinterface,  under device info it indicates it is connected to mqtt. 

I don't see anything published to mqtt, also otmonitor cannot connect ( connection refused).
I can connect to port 23, and I see messages going by.

This is device info:
```
Developer                                       Robert van den Breemen
NodeMCU Firmware Version          0.8.1+92f10cb (14-03-2021)
OTGW PIC Firmware Version         4.2.5
Compiled on (date/time)                 Mar 14 2021 14:56:26
HostName                                      OTGW
IP address                                     192.168.1.151
MAC address                                **
Free Heap Memory (bytes)          18328
Max. Free Memory (bytes)           16568
Unique Chip ID                             *****
Arduino Core Version                   2_7_4
Espressif SDK Version                 2.2.2-dev(38a443e)
CPU speed (MHz)                       160
Sketch Size (bytes)                      504.781
Free Sketch Space (bytes)         1540
Flash ID                                      00164020
Flash Chip Size (MB)                  4
Real Flash Chip Size (MB)         4
LittleFSsize                                 2
Flash Chip Speed (MHz)           40
Flash Mode                               DIO
Board Type                                ESP8266_NODEMCU
Wifi Network (SSID)                  ***
Wifi Receive Power (dB)            -55
MQTT Connected                     true
NTP Enable                              true
NTP Timezone                         Europe/Amsterdam
Uptime since boot                    0(d)-00:03(H:m)
Last Reset Reason                  Power On
Nr. Reboots                               3
```
I am a bit lost how to debug this.

rvdbreemen, 2021-03-14:
@DaveDavenport could you downgrade to version 0.8.0?

rvdbreemen, 2021-03-14:
@DaveDavenport also to what port are you connecting for OTmonitor?

DaveDavenport, 2021-03-14:
aah that works. did that change? 

I can see in my mqtt server logs it does connect to the server.
It also pushes the homeassistant settings (I don't use) 
```
homeassistant/sensor/otgw-2CF4327817F3/ASF_oemfaultcode/config
```

but nothing under `OTGW/#`

Ill try downgrade.

rvdbreemen, 2021-03-14:
@DaveDavenport yes it changed in v0.7.1

What version where you using before?  The way it works has changed in version v0.8.0. That has been an breaking change. The OTGW now published its topics under it's device. By making it unique it can actually work with multitple OTGW's on a single hub.

So what are you using as your home automation platform? Are you depending on the toptopic ('/OTGW'). If you explore your MQTT broker, you will notice that the published sensors are all still there, they just moved.

rvdbreemen, 2021-03-14:
The format for the mqtt topic is now:
- /OTGW/value/otgw-<uniqueid>/<topics>
- /OTGW/set/otgw-<uniqueid>/command/<command>

I just realized that I need to update the documentation to match.

rvdbreemen, 2021-03-14:
> aah that works. did that change?

> homeassistant/sensor/otgw-2CF4327817F3/ASF_oemfaultcode/config
So the auto discovery templates are pushed...

> but nothing under `OTGW/#`
That's weird, you should see values under the value topic, except that now you have the unique identifier in between.

> Ill try downgrade.
You have to drop to a release below v0.8.0 to not have the unique id's.

Please let me know if that worked for you.

DaveDavenport, 2021-03-14:
Yeah, I was many releases behind.
That and I saw the breaking change, so I did a complete clean flash.

As I mentioned I did not see anything on `OTGW/#'.  But I will look further down the chain.

rvdbreemen, 2021-03-14:
You should see topics published under /OTGW/ though. So I wonder what's going on. Could you verify v0.8.0 for me, I overlooked something in v0.8.1

DaveDavenport, 2021-03-14:
I feel a bit stupid now, it works..  (maybe I should stop doing this in the evening).... 

But I cannot explain it;  I saw that I was getting weird sockets errors in my mosquitto daemon.. realized was on an old version, so upgraded the daemon.. 
now it works.

Weird thing is I have everything going via mosquitto, and everything worked fine (my log files confirm this) except otgw.
I am confused af, but will take it as a win.. Sorry for wasting your time (again).

closed 2021-03-14 by DaveDavenport

rvdbreemen, 2021-03-14:
@DaveDavenport well no problem. Better you come and reporting an issue, than having a broken release. You are welcome to join the Discord to chat realtime next time. Just go here: https://discord.gg/gRVUQBKHbp

DaveDavenport, 2021-03-14:
I bought you a few beers! thanks.