rvdbreemen/OTGW-firmware #34: Changing settings needs reboot

PvdGulik, 2021-03-15:
After changing settings, I found out I have to reboot. Obviously, but is not mentioned. Maybe good suggestion?

rvdbreemen, 2021-03-15:
Do you know what settings did not work for you? I could even force a reboot, but so far I have not done that on purpose.

PvdGulik, 2021-03-15:
I'm not sure but it looked that the setting "MQTT OT msg Enable" does not respond.
But after reboot I found also no difference. So maybe it does something different.

What its supposed to do?

rvdbreemen, 2021-03-15:
The MQTT OT message enabled prevents the OTmessage topic to be published every second.

rvdbreemen, 2021-03-15:
@PvdGulik just checked the settings update code, to see what value would required updates. But only the sensor code might need an update after updating. All other fields, as far as I can tell only need an update. And should be in effect instantly.

closed 2021-03-15 by rvdbreemen