rvdbreemen/OTGW-firmware #35: Disable  un-used MQTT-messages

PvdGulik, 2021-03-15:
Is there a possibility to mark out all the MQTT topics you don't use/need?
E.g. my boiler does not support cooling, but OTGW Firmware does send out the MQTT messages:

```
> OTGW/value/otgw-ECFABC05B55B/cooling_enable
> OTGW/value/otgw-ECFABC05B55B/cooling
```

I tried to disable them in mqttha.cfg, but no succes

Thank you

rvdbreemen, 2021-03-15:
The reason they are send out is that they are part of the status byte message, I have no way of know if a system has them or not, they are just part of the protocol. 

I am not planning to leave them out at this point or making it configurable. Besides a bit of bandwidth waste.

closed 2021-03-15 by rvdbreemen

PvdGulik, 2021-03-22:
Hello Ruud,

I'd liked to reopen this issue, or maybe open a new issue.
Due to the large ammount of OTGW messages it looks like Domoticz is running out of memory which ends in a Domoticz Crash.
The problem is that slowly the memory increases, and, after 12 hours, a crash. This process cycles every 12 hours.

I use OTGW-Domoticz-MQTT-Client from proditaki to get the MQTT messages from OTGW Firmware via my MQTT-broker in Domoticz.

Yesterdag morning I stopped the OTGW MQTTClient hardware, and the memory issue stopped.

Proditaki asked me to address the issue here.

I think limiting the messages to only the one used by the specific boiler/thermostat will solve this problem.

Thank you

DaveDavenport, 2021-03-22:
Isn't this more of a Domoticz (or OTGW-Domoticz-MQTT-Client) issue?   reducing the amount of messages won't 'solve' domiticz bug, but just slow down the point time it takes to run out of memory?

PvdGulik, 2021-03-22:
It's only with this hardware. That's why I started with a issue at OTGW-Domoticz-MQTT-Client from proditaki.
He asked me to issue this here.

sjorsjuhmaniac, 2021-03-22:
@PvdGulik, I hate to tell you but this looks like a memory bug in Domoticz or the plugin as @DaveDavenport already mentioned. Even if the messages would be reduced it might not fix your issue but just postpone it. 

For now, you can reduce the amount messages by 
- unchecking the `MQTT OT msg Enable` option on the config page.
- going to the File explorer, downloading mqttha.cfg, removing all entities you don't want in Domoticz*, uploading it via file explorer with the original filename (overwriting the original) and rebooting the OTGW nodeMCU

*looking at how @proditaki implemented the plugin, he [is parsing the Home Assistant Discovery](https://github.com/proditaki/OTGW-Domoticz-MQTT-Client/blob/a352f864f9c5278626d6d76983453a9aa731b2fa/plugin.py#L132). So whatever is NOT in the HA autodiscovery, will not show up in your Domoticz and won't accelerate the Domoticz Memory Bug.

PvdGulik, 2021-03-25:
First I tried is the "unchecking the MQTT OT msg Enable option on the config page"
I'll report tomorrow the result.

Next step will editing mqttha.cfg.

Keep you informed.

PvdGulik, 2021-03-27:
Well, that didn't solve anything, same result ;-(
I'll try editing mqttha.cfg

PvdGulik, 2021-03-27:
Oke, I've edited mqttha.cfg. I disabled all with // at the beginning of the line.
Then only actived the ones I use and rebooted. Also in Domoticz all the devices created which I don't use are disabled.

After reboot Domoticz and OTGW NodeMCU, still 34 Sensors and Switch which I don't use in Domoticz are updated on regular basis. 15 of them are updated every 3 seconds.... Why?

These are: 
1345 | OTGW MQTT Client | 0027001A | 26 | OTGW MQTT Client - OTGW_Central_Heating | Light/Switch | Switch | Off | - | - |   | 2021-03-27 12:33:01
  |   | 1346 | OTGW MQTT Client | 0027001B | 27 | OTGW MQTT Client - OTGW_Domestic_Hot_Water | Light/Switch | Switch | Off | - | - |   | 2021-03-27 12:33:01
  |   | 1347 | OTGW MQTT Client | 0027001C | 28 | OTGW MQTT Client - OTGW_Flame | Light/Switch | Switch | Off | - | - |   | 2021-03-27 12:33:01
  |   | 1348 | OTGW MQTT Client | 0027001D | 29 | OTGW MQTT Client - OTGW_Cooling | Light/Switch | Switch | Off | - | - |   | 2021-03-27 12:33:01
  |   | 1349 | OTGW MQTT Client | 0027001E | 30 | OTGW MQTT Client - OTGW_Central_Heating_2 | Light/Switch | Switch | Off | - | - |   | 2021-03-27 12:33:01
  |   | 1350 | OTGW MQTT Client | 0027001F | 31 | OTGW MQTT Client - OTGW_Diagonostic_Indicator | Light/Switch | Switch | Off | - | - |   | 2021-03-27 12:33:01
  |   | 1413 | OTGW MQTT Client | 0027005E | 94 | OTGW MQTT Client - OTGW_Return_water_temperature | Temp | LaCrosse TX3 | 1.0 C | - | - |   | 2021-03-27 12:32:52
  |   | 1344 | OTGW MQTT Client | 00270019 | 25 | OTGW MQTT Client - OTGW_Fault | Light/Switch | Switch | Off | - | - |   | 2021-03-27 12:33:01
  |   | 1351 | OTGW MQTT Client | 00270020 | 32 | OTGW MQTT Client - OTGW_Central_Heating_enable | Light/Switch | Switch | Off | - | - |   | 2021-03-27 12:33:00
  |   | 1352 | OTGW MQTT Client | 00270021 | 33 | OTGW MQTT Client - OTGW_Domestic_Hot_Water_enable | Light/Switch | Switch | Off | - | - |   | 2021-03-27 12:33:00
  |   | 1353 | OTGW MQTT Client | 00270022 | 34 | OTGW MQTT Client - OTGW_Cooling_enable | Light/Switch | Switch | Off | - | - |   | 2021-03-27 12:33:00
  |   | 1354 | OTGW MQTT Client | 00270023 | 35 | OTGW MQTT Client - OTGW_OTC_enable | Light/Switch | Switch | Off | - | - |   | 2021-03-27 12:33:00
  |   | 1355 | OTGW MQTT Client | 00270024 | 36 | OTGW MQTT Client - OTGW_central_heating_2_enable | Light/Switch | Switch | Off | - | - |   | 2021-03-27 12:33:01
  |   | 1436 | OTGW MQTT Client | 00270075 | 117 | OTGW MQTT Client - OTGW_Status_Master | General | Custom Sensor | 0 # | - | - |   | 2021-03-27 12:33:00
  |   | 1437 | OTGW MQTT Client | 00270076 | 118 | OTGW MQTT Client - OTGW_Status_Slave | General | Custom Sensor | 0 # | - | - |   | 2021-03-27 12:33:01
  |   | 1410 | OTGW MQTT Client | 0027005B | 91 | CV water aanvoer (OTGW) | Temp | LaCrosse TX3 | 52.5 C | - | - |   | 2021-03-27 12:33:01
  |   | 1423 | OTGW MQTT Client | 00270068 | 104 | OTGW MQTT Client - OTGW_Relative_Modulation_Level | General | Percentage | 0% | - | - |   | 2021-03-27 12:32:58
  |   | 1406 | OTGW MQTT Client | 00270057 | 87 | OTGW_Control_setpoint | Temp | LaCrosse TX3 | 6.0 C | - | - |  

 I don't use them, are updated every 3 seconds. Why? And is this why I have a Domoticz_memory problem?

rvdbreemen, 2021-03-27:
I think @proditaki needs to look at the domoticz plug-in. As your memory issue is not cause by the messages itself. Reducing the number of message will only delay the problem. 

The root cause needs to be addressed. 

The sensors and switches are generated based on the specification.  As the setup is generic and standard based. It generates topics based in the messages that are being send. 

It’s also different for each thermostat and boiler. So making it less generic or more configurable over complicates the whole setup. 

The amount of messages and frequency should not be an issue, filtering can be done downstream. 

I push on change approach. The real problem with that is when you lose connectivity or values do not change for a long time. Than you could be out of sync.