rvdbreemen/OTGW-firmware #41: Upgrade PIC  Failed

joostvanmourik, 2021-03-19:
HI trying to upgrade the PIC trough the web interface, nothing happens and the telnet session displays this 👍 
13:35:43][ 17304| 15968] upgradepic (1251): Action: upgrade gateway.hex 5.1
[13:35:43][ 16584| 15968] upgradepicno(1150): Start PIC upgrade now.
[13:35:53][ 18296| 16304] fwupgradedon(1165): Upgrade finished: Errorcode = 9 - 10 retries, 0 errors
[13:35:54][ 18296| 16304] resetOTGW ( 65): Received firmware version: [?] [4.3] (3)
[13:35:54][ 16952| 16304] resetOTGW ( 72): Current firmware version: 4.3

This is what the logs shows on a attempt to upgrade

What am i doing wrong, and how to fix it ?

rvdbreemen, 2021-03-19:
Well, the errorcode 9 = too many retries. I have no idea why it failed, it complains about retry's. So for some reason it cannot write to the PIC. 

Do you really need to upgrade to 5.1? Most functionality just works with 4.3 already. The upgrade function has been integrated because people tried OTA upgrade of the PIC, which is a bad idea. However its now so simple that people upgrade because they can.

So do you have a need for a higher version?

closed 2021-03-19 by rvdbreemen

joostvanmourik, 2021-03-22:
HI @rvdbreemen not at this moment but i thought maybe you like to know the errors. 4.3 Seems to work fine for now, i can see all values in HA but i cannot alter the temperature in ha with the Climate instance. Is that by design because my Nest is stil the master ?

sjorsjuhmaniac, 2021-03-22:
It's definitely not by design. The equipment matrix shows that the Nest v3 OT doesn't support ID9 which is ..... remote room temp (setpoint) override 
http://otgw.tclcode.com/matrix.cgi#thermostats

joostvanmourik, 2021-03-23:
> It's definitely not by design. The equipment matrix shows that the Nest v3 OT doesn't support ID9 which is ..... remote room temp (setpoint) override
> http://otgw.tclcode.com/matrix.cgi#thermostats

Could it be something is wrong with the PIC (soldering) or shouldnt i see any information in that case.

sjorsjuhmaniac, 2021-03-23:
For not being able to change the thermostat setpoint: no.

For the flash error: i'm not an hardware expert. Note that there's a whole '[debugging](http://otgw.tclcode.com/debugging.html)' page over at otgw.tclcode.com that describes various things you can check.