rvdbreemen/OTGW-firmware #52: No updates in WebUI

glsf91, 2021-04-05:
I'am using 0.8.2+095b077 (25-03-2021) but the values in the Web UI are still not updated.
For example: Room Temperature Setpoint is still 20.5°C, even if I see other values in Domoticz if I change the setpoint on the thermostat.

rvdbreemen, 2021-04-05:
I thought we solved this in 0.8.2. Can you share some more detail? Does the binary sensor values change, like flame status? In the WebUI?

rvdbreemen, 2021-04-05:
The WebUI does update with my setup. So I wonder what the exactly your setup and steps to reproduce are. Could you share some debug logging, just telnet to port 23 and log for a while. Try changing the setpoint on your thermostat. And let me know what's going on.

glsf91, 2021-04-06:
Flame is also not changing. I don't have MQTT enabled.
Debug logging:

```
[11:37:14][  16616| 16112] handleOTGW  (1126): Net2Ser: Sending to OTGW: [SC=7:59/2] (9)
[11:37:14][  15272| 14816] handleOTGW  (1126): Net2Ser: Sending to OTGW: [PR=G] (4)
[11:37:14][  15272| 14816] handleOTGW  (1126): Net2Ser: Sending to OTGW: [PR=I] (4)
[11:37:14][  15272| 14816] handleOTGW  (1126): Net2Ser: Sending to OTGW: [PR=O] (4)
[11:37:14][  15232| 14672] handleOTGW  (1126): Net2Ser: Sending to OTGW: [PS=1] (4)
[11:37:14][  14032| 13520] checkOTGWcmd( 841): CmdQueue: Checking if command is in in queue [SC: 07:59/2] (11)
[11:37:14][  15744| 14816] sendOTmonito( 264): sending OT monitor values ...
[11:37:14][  13112| 12224] checkOTGWcmd( 841): CmdQueue: Checking if command is in in queue [PR: G=00] (8)
[11:37:14][  12440| 11576] checkOTGWcmd( 841): CmdQueue: Checking if command is in in queue [PR: I=00] (8)
[11:37:14][  12440| 11576] checkOTGWcmd( 841): CmdQueue: Checking if command is in in queue [PR: O=N] (7)
[11:37:14][  12440| 11576] checkOTGWcmd( 841): CmdQueue: Checking if command is in in queue [PS: 1] (5)
[11:37:14][  16512| 14512] sendOTmonito( 264): sending OT monitor values ...
[11:37:14][  12536| 11920] processOTGW (1076): Not processed, received from OTGW => [00000001/00001010,65.00,00000011/00000011,0.00,19/27,25.50,0.00,1.90,19.28,66.09,0.00,0.00,56.10,65/40,80/0,60.00,80.00,12663,3251,2722,81,1846,4190,7350,1] [155]
[11:37:14][  17184| 15720] sendOTmonito( 264): sending OT monitor values ...
[11:37:14][  17184| 15720] sendOTmonito( 264): sending OT monitor values ...
<removed same lines>
[11:37:32][  17472| 16312] sendOTmonito( 264): sending OT monitor values ...
[11:37:33][  17184| 16312] sendOTmonito( 264): sending OT monitor values ...
[11:37:44][  16552| 15912] handleOTGW  (1126): Net2Ser: Sending to OTGW: [SC=7:59/2] (9)
[11:37:44][  15368| 14616] handleOTGW  (1126): Net2Ser: Sending to OTGW: [PR=G] (4)
[11:37:44][  15368| 14616] handleOTGW  (1126): Net2Ser: Sending to OTGW: [PR=I] (4)
[11:37:44][  15368| 14616] handleOTGW  (1126): Net2Ser: Sending to OTGW: [PR=O] (4)
[11:37:44][  15328| 14616] handleOTGW  (1126): Net2Ser: Sending to OTGW: [PS=1] (4)
[11:37:44][  14128| 13320] checkOTGWcmd( 841): CmdQueue: Checking if command is in in queue [SC: 07:59/2] (11)
[11:37:44][  15840| 14616] sendOTmonito( 264): sending OT monitor values ...
[11:37:44][  13208| 12720] checkOTGWcmd( 841): CmdQueue: Checking if command is in in queue [PR: G=00] (8)
[11:37:44][  12536| 12072] checkOTGWcmd( 841): CmdQueue: Checking if command is in in queue [PR: I=00] (8)
[11:37:44][  12536| 12072] checkOTGWcmd( 841): CmdQueue: Checking if command is in in queue [PR: O=N] (7)
[11:37:44][  12536| 12072] checkOTGWcmd( 841): CmdQueue: Checking if command is in in queue [PS: 1] (5)
[11:37:44][  16512| 13776] sendOTmonito( 264): sending OT monitor values ...
[11:37:44][  12536| 11832] processOTGW (1076): Not processed, received from OTGW => [00000001/00000010,56.00,00000011/00000011,0.00,19/27,21.00,0.00,1.90,19.28,66.05,0.00,0.00,56.10,65/40,80/0,60.00,80.00,12663,3251,2722,81,1846,4190,7350,1] [155]
[11:37:44][  17184| 15072] sendOTmonito( 264): sending OT monitor values ...

```

But after refresh Web GUI is see also:

```
[11:40:56][  16416| 14712] handleDebug (  24): Wifi is connected
[11:40:56][  16416| 14712] handleDebug (  27): Reconnecting MQTT
[11:40:56][  16416| 14712] handleDebug (  33):
Debug MQTT: true
[11:40:56][  17088| 14712] handleDebug (  24): Wifi is connected
[11:40:56][  16416| 14712] handleDebug (  27): Reconnecting MQTT
[11:40:56][  16416| 14712] handleDebug (  24): Wifi is connected
[11:40:56][  16416| 14712] handleDebug (  27): Reconnecting MQTT
[11:40:56][  16416| 14712] handleDebug (  31):
Debug OTmsg: true
[11:40:56][  17088| 15360] handleDebug (  24): Wifi is connected
[11:40:56][  16416| 14712] handleDebug (  27): Reconnecting MQTT
[11:40:56][  16416| 14712] handleDebug (  32):
Debug RestAPI: true
[11:40:56][  16416| 14712] handleDebug (  24): Wifi is connected
[11:40:56][  16416| 14712] handleDebug (  27): Reconnecting MQTT
[11:40:56][  16416| 14712] handleDebug (  33):
Debug MQTT: false
[11:40:56][  16416| 14712] handleDebug (  12): Configure MQTT Discovery
[11:40:56][  16416| 14712] handleDebug (  13): Enable MQTT: false[11:40:56][  17224| 16312] operator()  ( 108): in 'onNotFound()'!! [/api/v1/otgw/otmonitor] =>
[11:40:56][  15880| 15016] operator()  ( 111): next: processAPI(/api/v1/otgw/otmonitor)
[11:40:56][  15880| 15016] processAPI  (  27): from[192.168.1.97] URI[/api/v1/otgw/otmonitor] method[GET]
[11:40:56][  15840| 15016] processAPI  (  43): >>word[0] => [], word[1] => [api], word[2] => [v1], word[3] => [otgw], word[4] => [otmonitor],
[11:40:56][  15840| 15016] sendOTmonito( 264): sending OT monitor values ...
[11:40:56][  17200| 15016] operator()  ( 108): in 'onNotFound()'!! [/api/v0/devtime] =>
[11:40:56][  16528| 15016] operator()  ( 111): next: processAPI(/api/v0/devtime)
[11:40:56][  16528| 15016] processAPI  (  27): from[192.168.1.97] URI[/api/v0/devtime] method[GET]
[11:40:56][  16504| 15016] processAPI  (  43): >>word[0] => [], word[1] => [api], word[2] => [v0], word[3] => [devtime],
[... 785 lines cut ...]
[11:52:55][  15880| 14792] processAPI  (  27): from[192.168.1.97] URI[/api/v1/otgw/otmonitor] method[GET]
[11:52:55][  15840| 14792] processAPI  (  43): >>word[0] => [], word[1] => [api], word[2] => [v1], word[3] => [otgw], word[4] => [otmonitor],
[... cut ...]
[11:52:57][  16280| 15440] operator()  ( 111): next: processAPI(/api/v1/otgw/otmonitor)
[11:52:57][  16280| 15440] processAPI  (  27): from[192.168.1.97] URI[/api/v1/otgw/otmonitor] method[GET]
[11:52:57][  16240| 15440] processAPI  (  43): >>word[0] => [], word[1] => [api], word[2] => [v1], word[3] => [otgw], word[4] => [otmonitor],
[11:52:57][  16240| 15440] sendOTmonito( 264): sending OT monitor values ...
[11:52:57][  17200| 15440] operator()  ( 108): in 'onNotFound()'!! [/api/v0/devtime] =>
[11:52:57][  16528| 15440] operator()  ( 111): next: processAPI(/api/v0/devtime)

```

rvdbreemen, 2021-04-06:
Did you update the LittleFS with the one from the repo?

glsf91, 2021-04-06:
No. I used https://github.com/rvdbreemen/OTGW-firmware/wiki/How-to-flash-the-binary-release-to-your-ESP8266-NodeMCU
I did not have a previous install on this nodemcu. It was a first time flash.

rvdbreemen, 2021-04-06:
Thanks for sharing the logs. I think I have an idea what is going on.

1. So it looks like you have updates correctly. If you get no error message in the webui on the left bottom the firmware and littlefs filesystem are in sync.
2. Also you use the serial2net connection from Domoticz to control the OTGW.
3. In the logs are you can see that Domoticz sends several commands, one of which is PS=1.
See here: [11:52:14][  15472| 14688] handleOTGW  (1126): Net2Ser: Sending to OTGW: [PS=1] (4)

This command Print State (PS) is explained here: 
https://otgw.tclcode.com/firmware.html#operation

This means that the raw updates of OpenTherm is stopped, and the functions of the OTGW firmware, is limited to being a serial2net proxy. All advanced functions like, MQTT topics, controlling using MQTT, the RESTapi and the WebUI all are based on the parsing on the Open Therm raw message processing.

In other words, it's correct that you do not see the WebUI being updated, as you use the Domoticz Climate control. 

Hope this answers your question,
Robert

glsf91, 2021-04-06:
Ok thanks.
Maybe add this to the wiki  (https://github.com/rvdbreemen/OTGW-firmware/wiki/How-to-use-it-with-Home-Assistant-or-Domoticz-or-others under Serial 2 network)?

It is also a  little confusing a page is showed with is not updated. Maybe better to not show this page because values are not right.

rvdbreemen, 2021-04-06:
I agree with you that the documentation needs updating. Your feedback is much appreciated on this issue. I will take an action to improve the documentation and consider a way to disable the webUI when moving to PS=1 mode.

glsf91, 2021-04-06:
Does this also mean there is no reason for my current use to use your firmware because it is not doing anything else then espeasy also can do (just een serial to net converter) ?
Only maybe that I can update the PIC firmware with your software.
Yes, I know there is another plugin for Domoticz :-)

rvdbreemen, 2021-04-06:
If you are happy with easyesp. Then by all means use that. 

Just know I this project to replace easyesp in the first place. I was having big issues with ser2net function. And it made a lot of sense to implement parsing on the esp so that I no longer need to depend on other solutions to create MQTT or to depend on a serial connection over WiFi. 

That’s how the project got started. It’s wat works for you best 😉

rvdbreemen, 2021-05-01:
Todo: Documentation updating explaining the classic integration effect. Next release will contain PS=1 detection.

rvdbreemen, 2021-05-02:
A small improvement in 0.8.4 with a detection of PS=1. Documentation updated.

closed 2021-05-02 by rvdbreemen