rvdbreemen/OTGW-firmware #54: GPIO Output Uses GPIO Sensors interval

sjorsjuhmaniac, 2021-04-11:
Right now 'output sensors' use the the GPIO Sensors Interval setting.
https://github.com/rvdbreemen/OTGW-firmware/blob/05df1bdbf92b30836071d51893bc7f50ceb85f9d/OTGW-firmware.ino#L38-L39

Although probably not a real issue, this can cause strange/unexpected behavior. 
I suggest to create a new interval var for the ouputs and remove the TODO line upon completion.

rvdbreemen, 2021-04-11:
I will check what it the correct lines of code. This needs cleaning up.

rvdbreemen, 2021-04-12:
Removed the interval completly. It should not be an constant, it should just follow the status bit. I moved the evaloutput() to background task. That way it will always be run, when enabled.

closed 2021-04-12 by rvdbreemen