rvdbreemen/OTGW-firmware #56: NodeMCU not accesible when mounted on OTgatewayboard

EdwinD007, 2021-04-20:
Hello Robert,
Very good you're busy with the OTGW firmware. I've used it for two week with pleasure. 
2 months ago I ran in some problems. I started when I wanted to upgrade the PIC to fw 5.1. I assumed I bricked the PIC as the Nodemcu works fine dismounted, but seems to reboot every 15 sec when mounted on the OTGW board. So first I sent the PIC to NODO and they were so kind to put fw 5.1 on it. But still no luck. Then I decided to buy a fresh OTGW board, which worked fine for 2 weeks with fw 5.1 and 0.8.3 running on homeassistant. But then it stopped again. And I have exact the same sympton as with the old board. Nodemcu of the board works fine. I put a FTDI cable on both boards and that seems also to work fine, when running OTmonitor. So the board with PIC seem okay. So then put the old fw on the Nodemcu (R147) and everything worked again, also mounted on the OTGW board. It looks like that something on the OTGW board triggers a reset. Is there a GPIO which has to be configured in the 0.8.3 firmware or something?  I read on the tweakers forum someone with same issue but didn't see a solution in the thread. 
NodeMCU Firmware Version0.8.3+2fa1f1f (05-04-2021)
OTGW PIC Firmware Version 5.1
Compiled on (date/time)Apr 5 2021 21:56:13
Already thanks for taking the time to read.
Edwin

rvdbreemen, 2021-04-20:
Hi Edwin,
Well, you are the first to report an issue like this. 

Did you try flashing a fresh copy of v0.8.3 bin files to your nodemcu? As you seem to have gone back to R147. 

Also do you have some debug log information? You can capture that with a telnet to port 23. 

If you enter a reboot loop, what do the two blue leds do? Do they blink, does it get stuck?

There is a watchdog reset that actually could cause an external reset. 

If you join the discord we can chat in real time. 

Do you have a link to tweakers about this issue?

Thank
Robert

EdwinD007, 2021-04-21:
Hi Robert,

Thanks for responding that quick and your help. 
Yes flashed the 2 files of v083. Also tried erase first, and then a fw flash. 
Good point about the telnet. I got some information;
================
[00:00:11][  16120| 15144] checklittlef( 591): Check githash = [2fa1f1f]
[00:00:11][  15080| 13848] initOutputs (  18): inside initOutputsO1...
[00:00:11][  15080| 13848] initWatchDog( 172): Setup Watchdog
[00:00:11][  15080| 13848] initWatchDog( 173): INIT : I2C
[00:00:12][  16424| 15144] initWatchDog( 191): INIT : Reset by WD!
[00:00:12][  15080| 13848] processOTGW (1111): Not processed, received from OTGW                                              => [Thermostat disconnected] [23]
=========================
When powering up the otgw board with nodemcu mounted, 
Both blue led's are on at startup
After 8 seconds nodemcu led (GPIO16) stays on and the led near the antenna(GPIO2) goes off, 
From then it goes in a 12 sec cycle: nodemculed goes off for less than a second and directly after that the led near the antenna goes on for less than a second.

When nodemcu is connected to usb (dismounted from board)
Both led on at startup
After 12 sec the led near the antenna goes off
14 sec later the other led goes also off.
8 seconds later the led near the antenna goes in 1 sec flash cycle (heartbeat?)

Link from someone with simular probs (WackoH) 
https://gathering.tweakers.net/forum/list_messages/1653967/21        (17th message from top)
Thanksf for the effort!
Edwin

rvdbreemen, 2021-04-21:
Hi Edwin,

Thank you for your patience... Let's keep debugging then.  So please read [the explain on the LEDs](https://github.com/rvdbreemen/OTGW-firmware/wiki/The-meaning-of-the-blue-leds---debug-setup-phase-and-normal-operation), 

As far as I can tell you have a correct behaviour when NOT connected to the OTGW hardware. Then it is disconnected it ends up blinking the LED1 (heartbeat/watchdog) with a cycle of about 1s. That is normal behaviour, and the LED2 should stay off as you have no incoming data at all. 

It also means you are not triggering any  OTGW related process other than the watchdog keeping it stable. 

Now the hardware connected situation seems to be wrong, it seems to reset when connected right away. Also the log part you send, tells me it's sending a message "Thermostat disconnected". So the OTGW PIC, does not see a thermostat connected ( it even sends the corresponding message ). But it does not explain the reset at that point, as I do not reset for not having something connected.

So what to do next:
1. Did you actually have the thermostat connected correctly, and the boiler? Sometime a lose wire is a valid issue.
2. Capture some logs in disconnected mode (without hardware) and capture a full log (best you can) from a connected situation.

And share your finding here. Or, in the discord server online: https://discord.gg/TcmTg6BT

rvdbreemen, 2021-04-21:
@EdwinD007 one more request: please link the direct message on the tweakers forum. As the place you pointed me back at was from 2014 or so, that cannot be related to my firmware, as it only exists since the pandemic ;-)

rvdbreemen, 2021-04-26:
@EdwinD007 And any progress?

EdwinD007, 2021-04-26:
Hoi Robert,
Ik ben nog niet verder. Ben momenteel weekje weg. Dus ga volgende week weer
verder. Ik snap er nog steeds niets van want het heb goed gewerkt. Laatste
testen waren overigens zonder boiler en thermostaat aangesloten.
Bedankt voor de support zover. Als ik meer debug info heb stuur ik even een
post.
Groetjes Edwin

rvdbreemen, 2021-05-02:
@EdwinD007 Any progress? I will close this, if you need more support, then just reopen the issue and we will continue debugging.

closed 2021-05-02 by rvdbreemen