rvdbreemen/OTGW-firmware #58: Settings loading forever

alucryd, 2021-05-22:
Hi, I've got an issue with the settings page. It was working fine at first, but ever since I activated MQTT, it won't load anymore and is stuck on "Loading...", any idea what's going on?
[image]
Tried brave and firefox, same result.

rvdbreemen, 2021-05-22:
That's weird. Have you rebooted and checked what happens?

alucryd, 2021-05-27:
Yes I have, I've also tried downloading the settings.ini file, and manually disabled mqtt before uploading it back but that didn't change anything so it may be unrelated after all. No reboot is bringing it back, not a huge deal since we have access to settings.ini, but kind of annoying still.

No error in the console either, it just prints `refreshSettings()`, and then nothing :/ If I call it myself, it returns undefined, so that may be why.

rvdbreemen, 2021-05-27:
To find out what's going on could you telnet to port 23 to get debug information.  

Please capture the debug log and share it. 

Are you able to connect to your MQTT hub using a tool like MQTT explorer (http://mqtt-explorer.com/)?

When you share the logs I can help debug your issue.

alucryd, 2021-05-27:
Here's the telnet log, I tried accessing the settings at the same time but nothing jumped out to me: https://paste.xinu.at/EDrHCbm9FVI66WRI/

I have since disabled MQTT and am relying on the serial plugin in Home Assistant which works fine. I can reenable it though if you think you can find out more with MQTT.

rvdbreemen, 2021-05-27:
This debug log looks just fine, but as you mention without MQTT. If you could try an MQTT configuration again, and then log at the same time. Than we will get the information on what is going on with the MQTT connection. It might just explain what is happening there.

rvdbreemen, 2021-06-17:
Closing this issue as we never found out what the issue was.

closed 2021-06-17 by rvdbreemen