rvdbreemen/OTGW-firmware #59: MQTT Command topic stops working after 3 weeks

MrSnuggels, 2021-06-10:
Revision 0.8.4+f39b68a (02-05-2021)

I post command (outside temperature) at a 1 minute interval through the command topic on MQTT. This works perfectly for about 18 days, after that the OTGW does not listen to the command topic anymore. Publishing it's own topics to the broker remains operational without issues. Sending commands through the network port with the opentherm gateway application also remains operational, so it appears to be in the MQTT handling.

rvdbreemen, 2021-06-17:
Hi @MrSnuggels that's a weird issue... after 18 days it needs a reset? I immediately suspect a bug in on of the libs I use, like the pubsub for MQTT or something. So are you saying I need to reset once a week?

rvdbreemen, 2021-06-30:
@MrSnuggels hmmm, just fixed a cmd queue bug that could have been related to your problem. The commands where due based on nos() not millis(), that might also cause your issue. However I have no other way than asking you to upgrade to v0.8.5 when released and find out.

rvdbreemen, 2021-06-30:
Closing this issue, expect it is related to issue #60 and that was fixed. If you keep seeing your issue @MrSnuggels then please let me know and reopen this issue. We can investigate that if that would be needed.

closed 2021-06-30 by rvdbreemen

MrSnuggels, 2021-07-04:
Just flashed the new firmware release, will let you know in a couple of weeks if this resolves the issue.

MrSnuggels, 2021-08-09:
Issue seems to be resolved indeed. Now at 36 days since boot and still responding to MQTT commands. Thank you very much.

rvdbreemen, 2021-08-09:
Thanks for the confirmation @MrSnuggels