rvdbreemen/OTGW-firmware #62: The connection between NodeMCU board and OpenTherm Adapter

LaughingLeeLife, 2021-07-21:
Hi Rvdbreemen,
Thanks for this project. I'm trying to connect the Bocsh Boiler and the HomeAssistant with this OTGW-firmware.
But I couldn't buy "the Nodoshop NodeMCU/ESP8266 version of the OpenTherm Gateway" from China. I'll diy an OT adapter and connect it to NodeMCU. 
Could you help to share the connection info between the NodeMCU and OT adapter?

I just find below port definitions in OTGW-firmware.h.
#define I2CSCL D1
#define I2CSDA D2
#define BUTTON D3
#define PICRST D5

Thanks,
Laughing

rvdbreemen, 2021-07-21:
Hi @LaughingLeeLife 

The firmware is designed for a specific PCB with a PIC that adds functionality to the OTGW. The PIC makes it possible to actually control, monitor and overrule the Thermostat. 

The original OTGW project cab be found here: https://otgw.tclcode.com/index.html

You can buy the OTGW from nodoshop: https://www.nodo-shop.nl/nl/opentherm-gateway/188-opentherm-gateway.html

I understand that you build your own OT interface. Could you explain to me what you build and what you are trying to do with this 8266 firmware?

The define you found is for the i2c communication to the hardware watchdog. Which is only part of the nodoshop pcb. 

Please provide some more context. You could join the discord to discuss what you are trying to do with more than just me.

rvdbreemen, 2021-07-21:
Join the discord using: https://discord.gg/zjW3ju7vGQ

LaughingLeeLife, 2021-07-22:
Hi Rvdbreemen,
Thanks for your reply. 
I'm sorry I couldn't open the link: https://discord.gg/zjW3ju7vGQ.

My hardware is: NodeMCU(esp8266) + OpenTherm Adapter.
NodeMCU: [https://github.com/nodemcu/nodemcu-devkit-v1.0](url)
OpenTherm Adapter: 
[image]
[image]

I want to use HomeAssistant to control the boiler automatically. So I need the OpenTherm Gateway Firmware with a simple OpenTherm Adapter. I don't have the thermostat. All of the temperature sensors has been integrated in my Home Assistant.

Is it possible to split the OTGW fireware into NodeMCU/ESP8266 without the PIC?

Thanks,
Laughing

rvdbreemen, 2021-08-09:
@LaughingLeeLife you asked me:  
"Is it possible to split the OTGW firmware into NodeMCU/ESP8266 without the PIC?"

I think so, but I have not plans to do that myself at this point. I see what you need, I think the firmware could be adapted to work with the OT Adaptor. So you are welcome for fork the code and modify it accordingly. In case you do, let me know

closed 2021-08-09 by rvdbreemen