rvdbreemen/OTGW-firmware #63: Feature Request - Add DHT22 support

Marcello1970, 2021-08-03:
it is possible to choose between a DS18B20 or a DHT22 on the GPIO connection. In addition to the temperature, you can also measure the humidity in the room. That would be great.

rvdbreemen, 2021-08-03:
@Marcello1970 interesting idea, that would require some more coding. I have no experience with a DHT22 and the DS18B20 was developed by @sjors a while back. I am sure it is something that could be build.

Roos-AID, 2022-02-05:
I looked at this when making changes for the MQTT HA autodiscovery for sensors.  The key difference is that DB18B20 class of devices is using onewire protocol, allowing multiple sensors on a GPIO pin.
The DHT22 requires a GPIO pin for each device individually.  
So, if we restrict it to one DHT22, I can see if I can implement this.