rvdbreemen/OTGW-firmware #64: Cannot set control setpoint to 0

katilius, 2021-08-09:
Currently using version 0.8.3. Previously I was able to send this request:
`POST http://otgw.local/api/v1/otgw/command/CS=0 

this would turn off boiler if nothing is using it.

Now it no longer works, I can send :

`POST http://otgw.local/api/v1/otgw/command/CS=1` 

and it updates fine, but 0 is ignored.

rvdbreemen, 2021-08-09:
There was a bug in pre-0.8.5 that sometimes causes messages to stay in the queue for a long time.

Cloud you try the lars test firmware and validate if that solves your issue. 

Thanks
Robert

katilius, 2021-08-09:
Just updated to `0.8.5+dc3dc14. It worked once, but then stopped working after that. Also tried sending `CS=-1`  but does not work.

rvdbreemen, 2021-08-09:
It sounds like a bug. So I will try to replicate it. Could you share some debug logs from your problem?

rvdbreemen, 2021-08-09:
I just tested the latest firmware, and the REST API just works fine.

Not sure what problem you are experiencing, so could you please share some debug logs.
Read this: https://github.com/rvdbreemen/OTGW-firmware/wiki/How-to-debug-the-OTGW-firmware

If you want some realtime interaction you can share the Discord community. And chat about your issue with me and others. Just go here: https://discord.gg/zjW3ju7vGQ

rvdbreemen, 2021-08-09:
So did you read this part of the OTGW commands?

"CS=temperature
Control Setpoint - Manipulate the control setpoint being sent to the boiler. Set to 0 to pass along the value specified by the thermostat. To stop the boiler heating the house, set the control setpoint to some low value and clear the CH enable bit using the CH command.
Warning: manipulating these values may severely impact the control algorithm of the thermostat, which may cause it to start heating much too early or too aggressively when it is actually in control.
Example: CS=45.8, CS=0

C2=temperature
Control Setpoint for 2nd CH circuit - Manipulate the control setpoint being sent to the boiler for the second CH circuit. Set to 0 to pass along the value specified by the thermostat.
Example: C2=45.8, C2=0

CH=state
Central Heating - When using external control of the control setpoint (via a CS command with a value other than 0), the gateway sends a CH enable bit in MsgID 0 that is controlled using the CH command. Initially, this bit is set to 1. When external control of the control setpoint is disabled (CS=0), the CH enable bit is controlled by the thermos"

So what are the commands you are sending to the OTGW using the REST command interface?

rvdbreemen, 2021-08-09:
@katilius I noticed this: 
"To stop the boiler heating the house, set the control setpoint to some low value and clear the CH enable bit using the CH command." 

If I understand correctly, then you need to clear the CH bit and you need to set the CS to a low value.... 

Meaning to turn off your boiler, you send two commands back to back:
CS=0
CH=0

That should turn off the boiler then.

Could you try that?

katilius, 2021-08-09:
Yes, I missed the part about CH. Will try it now. Initially I was experimenting with all commands that my boiler supported and looked what could stop the boiler and using just CS worked up until I upgraded to version 0.8.

rvdbreemen, 2021-08-09:
@katilius did you upgrade you PIC to a newer version by any chance? As the 8266 firmware basically forward the commands

katilius, 2021-08-09:
I did, but I can double check. Is version somwehere available to check ?

rvdbreemen, 2021-08-09:
Goto the webUI, click Device Info. And check the "OTGW PIC Firmware Version". The current version I run is version 4.3 btw...

katilius, 2021-08-10:
I have tried sending CH as well, but now I am in different state, I cannot reenable heating. I send this request:
`http://otgw.local/api/v1/otgw/command/CH=1` 

But it does not turn back on.

Here is the log after sending command:
```
[07:29:58][  14408| 13640] addOTWGcmdto( 991): CmdQueue: Adding cmd end of queue, slot [0]
[07:29:58][  13064| 12344] addOTWGcmdto( 994): CmdQueue: Insert queue in slot[0]:cmd[CH=1] (4)
[07:29:58][  13064| 12344] addOTWGcmdto(1012): CmdQueue: Next free queue slot: [1]
[07:29:58][  12680| 11696] processOTGW (1163): Boiler           [B40193966] [READ_ACK        ]  [Tboiler                       ]        Tboiler                               = 57.40 °C
[07:29:59][  14248| 13096] handleOTGWqu(1024): CmdQueue: Checking due in queue slot[0]:[23727393]=>[23727776]
[07:29:59][  12904| 11800] processOTGW (1172): Request Boiler   [R80010000] [READ_DATA       ]  [TSet                          ]                hb[  0] lb[  0]
[07:29:59][  12904| 11800] processOTGW (1163): Boiler           [B70010000] [UNKNOWN_DATA_ID ]  [TSet                          ]                hb[  0] lb[  0]
[07:30:00][  14248| 13096] handleOTGWqu(1024): CmdQueue: Checking due in queue slot[0]:[23728393]=>[23727776]
[07:30:00][  12904| 11800] handleOTGWqu(1026): CmdQueue: Queue slot [0] due
[07:30:00][  12904| 11800] sendOTGW    (1108): Sending to Serial [CH=1] (4)
[07:30:00][  11560| 10504] checkOTGWcmd(1064): CmdQueue: Checking if command is in in queue [CH: 1] (5)
[07:30:00][  11560| 10504] checkOTGWcmd(1075): CmdQueue: Checking [CH]==>[0]:[CH=1] from queue
[07:30:00][  11560| 10504] checkOTGWcmd(1078): CmdQueue: Found cmd [CH]==>[0]:[CH=1]
[07:30:00][  11560| 10504] checkOTGWcmd(1081): CmdQueue: Found value [ 1]==>[0]:[CH=1]
[07:30:00][  10888|  9856] checkOTGWcmd(1082): CmdQueue: Remove from queue [0]:[CH=1] from queue
[07:30:00][  10888|  9856] checkOTGWcmd(1084): CmdQueue: Moving [1] => [0]
[07:30:00][  11920| 11400] processOTGW (1172): Request Boiler   [R00740000] [READ_DATA       ]  [BurnerStarts                  ]                hb[  0] lb[  0]
[07:30:00][  12904| 11800] processOTGW (1163): Boiler           [B407477E5] [READ_ACK        ]  [BurnerStarts                  ]        BurnerStarts                          = 30693
[07:30:01][  10664|  9936] processOTGW (1172): Request Boiler   [R00110000] [READ_DATA       ]  [RelModLevel                   ]                hb[  0] lb[  0]

```

rvdbreemen, 2021-08-10:
@katilius it in the end depends on your thermostat and boiler, not the firmware of the NodeMCU and the PIC. So you have to experiment with the settings, you have to probably do a combination of CH and CS to turn off and turn on your boiler. 

Just come to discord channel to discuss with others, or, if you are more comfortable with a forum you could go here: https://www.domoticaforum.eu/viewforum.php?f=75

rvdbreemen, 2021-08-10:
Closing this issue, because it's not a problem with the firmware it seems, but more on how to control your boiler.

closed 2021-08-10 by rvdbreemen