rvdbreemen/OTGW-firmware #65: ESP8266 - PIC connection issue

raf1000, 2021-08-15:
Hello 

I have managed to compile files and uploaded ESP8266 using Arduino IDE.

I can login into otgw.local. When I click **PIC firmware** nothing is displayed, and info “ No version found”.
When I click **Device Info** I can see data among which “Pic connected = false”

Hence I have uploaded PIC with  getway.hex using  otmonitor connected to OTGW gateway via FDDI/USB.
I can connect from otmonitor to OTGW gateway via FDDI and send a command like AA=70 , I can see response.
However while otmonitor is connected to OTGW gateway via TCP/IP, when  send command AA=70, no response.
Seems like ESP8266 is not connected with PIC☹. Any hitns how to check what is not working.
PS. I am abale to connect to OTGW monitor via telnet as well but not sure what to debug.

rvdbreemen, 2021-08-16:
Hi

You have already debugged a lot. It sounds like a connection problem. 

What version of the pcb do you have? If it's a version 2.x then you have jumpers you need to set correctly so that the pic and the NodeMCU are connected. 

Please let me know your findings,

Robert

raf1000, 2021-08-16:
The PCB  V2.0. I have changed TxP/RxP jumpers for ESP/LAN as per instruction (so they are closer to ESP8266 Antena). I bought  the gateway in April 2021 and now trying to make it running. I wanted to connect to Renovent Brink HR (Ventilation) which uses Opentherm protocol  (I am in touch with a person who managed to do it). But I have the base issue with connection:-( . PCB looks OK not sure if it might be something on the board? if I can check any pcb connections? Or contact nodo shop for hints?

tjfsteele, 2021-08-16:
That's very strange. What do you get when you connect to the NodeMCU via the web interface?

rvdbreemen, 2021-08-16:
@raf1000 when you goto the debug port (telnet to 23) then you should see live messages being decoded, as well as some debug information like sending MQTT messages. 

If you do not see anything there, then I agree with @tjfsteele it strange. The green led blinking from the PIC is what you expect. So if you do not see any commands being parsed, and the serial port OTmonitor port, does not respond either. That's why I suggested to check the jumpers, could you share a picture of how you have setup your jumpers at this point?

On the discord there are more people that could think along with you figuring out what the problem is.

And yes, if nothing else works, you can also just contact Nodoshop. He is always very helpful getting hardware running.

raf1000, 2021-08-16:
1) When using FDDI/USB, after clicking Gateway in otmanager, I have the following (green LED flashing):
18:34:24.936547	Command: PR=M
18:34:24.969383	PR: M=M
18:34:29.454368	Command: GW=1
18:34:29.471898	GW: 1
18:34:29.502723	R00000000	Read-Data 	Status (MsgID=0): 00000000 00000000
		   - CH enable: disabled (0)
		   - DHW enable: disabled (0)
		   - Cooling enable: disabled (0)
		   - OTC active: not active (0)
		   - CH2 enable: disabled (0)
		   - Summer/winter mode: winter (0)
		   - DHW blocking: unblocked (0)
18:34:30.465244	R00000000	Read-Data 	Status (MsgID=0): 00000000 00000000
		

2) When using NodeMcu, after clicking Gateway in otmanager, I have the following (green LED flashing):
18:38:32.725177	Command: PR=M
18:38:35.403822	Command: GW=1
	
When I log in to the otgw.loca, I have the following:
DeveloperRobert van den Breemen
NodeMCU Firmware Version0.8.5+dc3dc14 (01-07-2021)
OTGW PIC Firmware VersionNo version found
Compiled on (date/time)Aug 13 2021 22:12:24
HostNameOTGW
IP address192.XXX.X.XXX <correct IP address is here>
MAC addressE8:DB:84:DD:17:FA
Free Heap Memory (bytes)9472
Max. Free Memory (bytes)7096
Unique Chip IDdd17fa
Arduino Core Version3.0.1
Espressif SDK Version2.2.2-dev(38a443e)
CPU speed (MHz)160
Sketch Size (bytes)522.719
Free Sketch Space (bytes)1524
Flash ID001640D8
Flash Chip Size (MB)4
Real Flash Chip Size (MB)4
LittleFSsize2
Flash Chip Speed (MHz)40
Flash ModeDIO
Board TypeESP8266_NODEMCU
Wifi Network (SSID)<correct name of Wifi Network is here>
Wifi Receive Power (dB)-73
NTP Enabletrue
NTP TimezoneEurope/Amsterdam
Uptime since boot0(d)-00:02(H:m)
Last Reset ReasonExternal System
Nr. Reboots54
MQTT Connectedfalse
Thermostate connectedfalse
Boiler connectedfalse
Pic connectedfalse

raf1000, 2021-08-16:
Connected with telnet (I am not using MQTT here):
help

---===[ Debug Help Menu ]===---
1) Toggle verbose debug logging - OT message parsing
2) Toggle verbose debug logging - API handeling
3) Toggle verbose debug logging - MQTT module
q) Force read settings
m) Force MQTT discovery
r) Reconnect wifi, telnet, otgwstream and mqtt

[19:04:51][  11352|  9896] handleDebug (  74): MyDEBUG =true
[19:04:52][  12856| 10544] do5minevent ( 167): Uptime seconds: 677

rvdbreemen, 2021-08-16:
@raf1000 From the logs you have shared here I can say that there is no OT msg see at all. 

Log 1&2 are the same, the status message is a fake decoding I think. 

From the NodeMCU dev info I can see that the OTGW PIC does not see the boiler nor thermostat. 

You could try to connect the wires again. Just to see if that makes any difference. 

As far as I can see it's a "not connected" to the  OT bus, you should see an average of 2 msg per second.

tjfsteele, 2021-08-16:
The strange thing is that it seems to work when using the FTDI serial cable.

rvdbreemen, 2021-08-17:
@tjfsteele it seems he can program the PIC, but I have not seen any data in either log from the boiler or thermostat. So either the jumpers are not connected correctly, or, the wires to the PCB from the boiler and thermostat are not connected correctly.

If both are good, then I am thinking a problem with this PCB all together.

What about you?

tjfsteele, 2021-08-17:
In his original post he says

> I do not have boiler nor thermostat so can not do more tests

Could be relevant?

Tim

raf1000, 2021-08-17:
Yes, at the moment I do not have connected boiler nor termostat. However my understanding of the gateway is that it shall resposne on the command I am sending, confirming that gateway received the request.  Hence whileusing serial I have confirmation as follows:
18:34:24.936547 Command: PR=M
18:34:24.969383 PR: M=M
18:34:29.454368 Command: GW=1
18:34:29.471898 GW: 1

And when using NodeMCU there is no confirmation that Gateway received a command:
18:38:32.725177 Command: PR=M
18:38:35.403822 Command: GW=1

Also the information in the OTGW firmware:
OTGW PIC Firmware VersionNo version found
Pic connected false

led me to conclusion that there is no communication between PIC and NodeMcu

tjfsteele, 2021-08-17:
The jumpers should look like this:

[image]

raf1000, 2021-08-17:
[image]
Yes exacly, for NodeMCU I am changing jumpers as per my photo. When I use serial I need to change them agin

tjfsteele, 2021-08-17:
If you have a multimeter or continuity tester I would disconnect the board from power then check there is a path from RX and TX on the NodeMCU to the jumper caps, although I've never heard of this being a problem before.

I assume the NodeMCU does not have anything plugged in to its USB port when you are trying to use it with OTGW.

Tim

tjfsteele, 2021-08-17:
I would disconnect the board from power then use a meter to check for continuity between the jumper caps and the vias as shown in the photo.

Also carefully inspect the solder joints around the jumpers and around the level converters at the top of the picture.
[image]

rvdbreemen, 2021-08-17:
"OTGW PIC Firmware VersionNo version found
Pic connected false"

There is room for improvement in UI and software tests. 

The firmware "no version" happens with some versions of firmware more often. I have found resetting it multiple times this sometimes resolves it. 

The PIC connection false  should be improved based on your feedback. It is used to let Home Assisant know about the status of the OTGW as a whole. So the name is deceptive. 

Even when everything is offline you can still talk to the PIC. I think you describe the way to test it. 

Using the REST API you can easily test the command sending. It even validates the return commands for you. Just check the wiki on how to use the REST API. 

Sorry, for the confusion the UI causes.

raf1000, 2021-08-17:
When using power supplier I am not connecting to USB.
I have checked RX between green circles and Tx between red circles and both are OK/continuity (0 ohm).  
When I check between TxP and Tx on NodeMCu there is 22.8 kohm (probably some change of voltage on smd resistor?)

[image]

tjfsteele, 2021-08-17:
RxP and TxP are not directly connected to Rx and Tx on the NodeMCU, they go via level converters as the PIC works at 5V and the ESP works at 3.3V.

The photo you posted isn't very clear unfortunately when zoomed it, it's difficult to take a sharp macro photo so I don't blame you at all. As far as I can see there isn't a problem with the level converters which are the two transistors on the left.

I took a picture with my macro camera to show what it should look like. The transistor markings should all be the same.

[image]

raf1000, 2021-08-17:
The above was usuefull. I was checking connection paths of NodeMcu Rx - TxP and NodeMcu Tx - RxP
I can not check transistors but after checking all connections seems there is no connection between  NodeMcu Rx and 10K resistor as on the picture below
[image]

I could try to connect via cable but I have never slodered SMD (they are so small..) :-(
 I think I shall submit a complaint to nodo shop

rvdbreemen, 2021-08-17:
Okay, so this has turned out to be a hardware issue. 

Please do contact nodoshop. I will close the issue know, as it's not a software related issue 😉

Please let us know when it gets resolved by Nodoshop.

tjfsteele, 2021-08-17:
[image]

The pads marked in red should be connected together by tracks on the board.

Could the issue be the soldering of the NodeMCU socket?

raf1000, 2021-08-17:
Bingo. Thank you for indicating RXD! on the PCB. As the RXD -R10 connection is OK, I can wire RXD with Node MCU Rx and it works!
Regarding soldering I might have overhited and  but there is a connection from NodeMCU top of the PCB to  the bottom of PCB (where soldering look ok, nothing strange)
Again thank you for your help!

tjfsteele, 2021-08-17:
That's excellent news, thank you for letting me know!

rvdbreemen, 2021-08-17:
Great news. Thanks @tjfsteele for helping out.

bmaehr, 2021-11-19:
> [image] Yes exacly, for NodeMCU I am changing jumpers as per my photo. When I use serial I need to change them agin

The jumpers set to FTDI have been the problem for my why I got "No version found" and also flashing the firmware didn't work.  Please upgrade the documentation or better add a hint to the UI.

rvdbreemen, 2021-11-19:
@tjfsteele do you think the documentation needs to be clearified?

bmaehr, 2021-11-20:
> @tjfsteele do you think the documentation needs to be clearified?

The problem is that a user like me (bougth the OTGW and the NodeMCU at nodo-shop) has to first find out what the real problem is. 

With some searching after I got the packege today I found the wonderful landing page https://github.com/rvdbreemen/OTGW-firmware/wiki/ where all steps are described really well. Only the small step changing the jumpers on the OTGW is not there. And because the NodeMCU just shows "no verison found" I automatically expect, that the firmware on the OTGW is missing and not, that the communication with the OTGW was not working. So I was searching for instructions how to flash the firmware with NodeMCU (and tried the PIC firmware page without success).

rvdbreemen, 2021-11-20:
@bmaehr that clarifies where additional hints can be added. 

I think you just mean that when you get the message “no version found” there should be instructions to check the jumper setting. 

Did you use the install pdf from the nodo shop site at all?

tjfsteele, 2021-11-20:
I'm wondering if this is a conflict between the v2.3 manual and an older
board?

Hopefully the manuals in the shop will be labelled more clearly soon.

Tim

bmaehr, 2021-11-20:
> @bmaehr that clarifies where additional hints can be added.
>
> I think you just mean that when you get the message “no version found” there should be instructions to check the jumper setting.

That is my suggestion.

> To clarify one more thing: Did you use the install pdf from the nodo shop site at all?

In fact I did look at all documents from them and not one was really helpfull. I didn't need a document how to solder the board (didn't see there something about jumper either), I didn't have a Ethernet and didn't use a USB interface (even don't have a USB interface on the OTGW and didn't bought an Ethernet module) and additionaly the OTGW did look different than the one I got and also the NodeMCU look different and has an other name now (I have, as far as I can see, the versions form the picture above).

rvdbreemen, 2021-11-21:
When I look at the picture you shared earlier, I think the issue is caused by the fact that we the manual at the nodoshop site is currently for PCB version 2.3. After talking to Tim (hardware designer of the current nodoshop PCB) we decided to publish the hardware manual for the older hardware. The jumpers are in the pre-v2.3 designs of the hardware. 

We hope this solves the problem if anyone else encounters this, plus that the manuals stay available that way. Plus we have notificed Remco (owner of Nodoshop) about this. Asked him if he would be so kind to publish older versions of the manual, so people can find it when needed.

rvdbreemen, 2021-11-21:
Closing issue, as the documentation is now published in Discord #Hardware channel. I am considering adding additional hint in the wiki as well.

closed 2021-11-21 by rvdbreemen