rvdbreemen/OTGW-firmware #70: Mqtt domoticz

antonlamers, 2021-11-08:
Hi,

I did fill by MQTT Home Assistant prefix: domoticz. 
when I fill it with domoticz/in some messages are not updated. What is right to do. I can’t use the otgw mqtt plug-in because I have an user and password.

anton

rvdbreemen, 2021-11-08:
@antonlamers thanks for reporting. Can you be more specific on what topics do not update after the change in topic. 

And did you reboot the OTGW? Just to make sure that works.

antonlamers, 2021-11-08:
Hi,
No i didn’t reboot.
This is the difference between domoticz and domoticz/in by the prefix. Do I it the right way? I send you 3 pics from domoticz and one from the settings.
[cid:418172BD-6EB2-46F8-BA0B-9DDF85253603-L0-001][cid:6631D47A-CFEF-455F-B811-7BB826F93BB8-L0-001][cid:6CFF1982-29D5-40A2-8183-9693915606A9-L0-001][cid:A195C2CE-75FF-4061-9ACC-754F63521090-L0-001]

mvg,
Anton Lamers
Verstuurd vanaf mijn iPhone

rvdbreemen, 2021-11-08:
@antonlamers images do not work like that in GitHub. You can upload them if you like or just join the discord channel (https://discord.gg/zjW3ju7vGQ).

antonlamers, 2021-11-08:
Ze staan op discord

mvg,
Anton Lamers
Verstuurd vanaf mijn iPhone

rvdbreemen, 2021-11-08:
Screenshots gezien. Wat gaat er fout? In principe worden sensors ge-update op basis van MQTT en MQTT updates gebeuren als er berichten binnen komen van de thermostaat of ketel. 

Kan je uitleggen wat er fout gaat?

rvdbreemen, 2021-11-09:
Solved by downgrading PIC to 4.3. This seems to be the only way to get the Domoticz default plugin to work correctly. If you can, then please use the specific plugin for this.

closed 2021-11-09 by rvdbreemen