rvdbreemen/OTGW-firmware #71: Ds18b20 vs firmware 0.9.0

pro2call-nl, 2021-11-09:
I have 2 dallas temperature sensors connected to gpio10. After upgrading to 0.9.0 the sensors don't show in mqtt.

I downgraded the .ino file to 0.8.6 and there working again.

pro2call-nl, 2021-11-14:
the problem has been found.

1. i installed the new INO file
2. i installed the new Littlefs file.

I assumed an reboot has been done automaticly. 

after hitting the reboot button in the littlefs webpage the device connection was lost. 1 power reset and everything was working again.

closed 2021-11-14 by pro2call-nl

rvdbreemen, 2021-11-14:
Ahhh, that's weird, because after a flash the esp is rebooted... so why did it initially not bring back everything... but you fixed it by turning it off and on again.