rvdbreemen/OTGW-firmware #73: Newly flashed device stuck in boot loop

reyapo, 2021-11-12:
Hi,

To update my gateway from a serial to a network connection I purchased a [NodeMCU](https://www.antratek.nl/nodemcu-v2-lua-based-esp8266-development-kit) and flashed it according to the wiki. Unfortunately it immediately gets stuck in a boot loop. If I connect to the serial output it constantly outputs the following on a loop:
```
Booting....[0.9.0+21bea01 (06-11-2021)]

�$l�H2�xn�GW=R
[OTGW firmwa�O4d��GW=R
[OTGW firmwar
             |4`,l$$p0t�W=R
[OTGW firmwar�O,`l$l`:iGW=R
[OTGW firmware�pYbfrzD>�GW=R
[OTGW firmware�O,hl,p�(GW=R
[OTGW firmware 0l�CGS{D>�
                         Htxv`lld$pt�W=R
[OTGW firmware 2$l��l,x�GW=R
[OTGW firmware t4`,l$$pt�W=R
[OTGW firmware -�$l�Hl$xo�GW=R
[OTGW firmware -t|4`,l$4b:IGW=R
[OTGW firmware - |4`,l$4b8IGW=R
[OTGW firmware - Nodoshop version]

Booting....[0.9.0+21bea01 (06-11-2021)]

Last reset reason: [External System]
Attempting to connect to WiFi network
Start Wifi ...
AutoConnect to: EO,`l$4b8�W=R
[OTGW firmware - N��`l$4b:IGW=R
[OTGW firmware - NoH�WAMME@>�GW=R
[OTGW firmware - No�l�Hl,HZGW=R
[OTGW firmware - NodHtxwamldH�t�W=R
```

The `OTGW firmware` lines are printed relatively slowly (1 line every ~500ms). Sometimes the `Attempting to connect` lines appear on a loop as well. It does not matter if the device is placed in the gateway or only connected to the USB. I tried the following things:

- Flash using esptooly.py
- Flash using esptool-ck
- Older firmware versions (down to 0.8.0)
- Compile the firmware and flash with Arduino IDE

All produce the exact same behavior, which would lead me to the conclusion that I received a defective board. However If I flash [an example program](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/WiFiAccessPoint/WiFiAccessPoint.ino) using the Arduino IDE, it works immediately. 

Is there anything I can try? Did I do something wrong, or did I get a wrong version of the NodeMCU?

Please let me know if more info is needed. Thanks in advance!

rvdbreemen, 2021-11-13:
Hi @reyapo thanks for reporting your issue. I totally don’t understand why the device would not work unless it’s a bad clone or fault nodemcu. 

It indeed looks like a deep bootloop. The bootbanner is mostly seen in your logging on the usb. You hardly seem to get to the next step. 

Stupid question: is the nodemcu in the OTGW board when connected to the USB serial or not?

Cheers,
Robert

reyapo, 2021-11-13:
Hi Robert

Thanks for your quick reply. As far as I can tell the board is an original NodeMCU v1.2. 

The log I pasted above is when the board is connected through USB only, so not placed on the gateway. However if I place the board in the gateway the LEDs blink in the same way and no Wifi AP appears so I assumed it does the same. I did not read the serial at that time though. 

I ordered a second board, from the NODO shop this time, as a backup. Please let me know If you would like me to test certain things, otherwise I will wait for the second board and hope for better results with that one.

Thanks!

rvdbreemen, 2021-11-14:
@reyapo well, when the nodemcu comes please report here what happened. Thanks... I leave your issue open for now.

reyapo, 2021-11-16:
@rvdbreemen I got the new board and it worked immediately. Its now running and sending data to Home Assistant flawlessly! Why the other one is not working I don't know. If I put them side by side there are definitely differences, but unless someone else encounters the same issue I would say, call it defective and close the issue.

rvdbreemen, 2021-11-16:
Thanks for letting me know. Clossing issue.

closed 2021-11-16 by rvdbreemen