rvdbreemen/OTGW-firmware #74: Building with arduino-cli, missing OTGW-firmware.mklittlefs.bin

jvwdev, 2021-11-18:
I'm trying to build with arduino-cli, which works fine after installing some libs which are not added as git submodules yet.
After building there is a /tmp/arduino-sketch-* folder, which contains the OTGW-firmware.bin.

But there is no OTGW-firmware.mklittlefs.bin file to be found, how can I build this file ? 

Some details on how I setup and build with arduino-cli, which might be useful for the wiki.
I removed the esp8266 git submodule as I use the esp8266 url, just like in the arduino GUI.

```
# setup
git submodule sync --recursive
git submodule update --init --recursive --progress

arduino-cli config init --additional-urls https://arduino.esp8266.com/stable/package_esp8266com_index.json
arduino-cli core update-index
arduino-cli core install esp8266:esp8266
arduino-cli lib install acetime time

# build
arduino-cli compile --clean --fqbn esp8266:esp8266:d1_mini --libraries Arduino/libraries --verbose
```

rvdbreemen, 2021-11-18:
@jvwdev Thanks for the feedback. 

I have not tried building it using Adruino CLI myself. I personally use the VSC plus Adruino IDE plugin. 

So your input is more than welcome. Hope you will share all information on your buildpipeline once you got it done. 

In the issues you found:
1. Missing git submodules. You are right, I changes libraries, but forgot to fix that part. 

You can drop eztime library. Then you add acetime and timelib. 

2. Filesystem building
I use a FS plug-in to generate the littleFS filesystem bin. 

This is the GitHub location: 

https://github.com/earlephilhower/arduino-esp8266littlefs-plugin

Hope you can make it work and like I mentioned. When you do. Let me know. I will add it to the wiki. 

Robert

jvwdev, 2021-11-18:
Thanks for the info!

I looked at the plugin, it calls a tool called mklittlefs and has some logic to flash it.
The mklittlefs binary is provided by the esp8266 package. 

I made a quick shell script to make the image, which seems to be working.
The only problem is that the values as page/block-size and start/end addresses are hardcoded for now, as I'm not sure where to properly get these values. I looked at the OTGW-firmware.ino.map file.

Can you verify if the provided output of the mklittlefs looks similar as the plugin ?
```
SPI page   256
SPI block  8192
SPI start  1077936128
SPI end    1080008704
SPI size   2072576
/index.css
/gateway-5.0.ver
/FSexplorer.css
/diagnose.hex
/settings.ini
/version.hash
/FSexplorer.png
/gateway.ver
/interface.ver
/mqttha.cfg
/index.html
/refresh-page-option.png
/amica-nodemcu-esp8266-pinout.png
/FSexplorer.html
/transfer.dat
/diagnose.ver
/gateway.hex
/index.js
/gateway-4.3.ver
lfs debug:1469: Expanding superblock at rev 17
/settings.png
/favicon.ico
/gateway-5.0.hex
/interface.hex
/gateway-4.3.hex
/download-to-storage-drive.png
```

My build script fow now:
```
#!/bin/sh

set -e
set -u

IMAGE=/tmp/arduino-sketch-*/OTGW-firmware.ino.bin
IMAGE_LITTLEFS=OTGW-firmware.mklittlefs.bin
MKLITTLEFS=$HOME/.arduino*/packages/esp8266/tools/mklittlefs/*/mklittlefs
DATAPATH=data

# build OTGW-firmware.bin
arduino-cli compile --clean --fqbn esp8266:esp8266:d1_mini --libraries Arduino/libraries --verbose
cp $IMAGE OTGW-firmware.bin

# SPI values found in OTGW-firmware.ino.map
# TODO fetch the values automatically
# convert hex to dec
SPI_PAGE=$(echo $((16#100)))
SPI_BLOCK=$(echo $((16#2000)))
SPI_START=$(echo $((16#40400000)))
SPI_END=$(echo $((16#405fa000)))
SPI_SIZE=$(expr $SPI_END - $SPI_START)

echo "SPI page   $SPI_PAGE"
echo "SPI block  $SPI_BLOCK"
echo "SPI start  $SPI_START"
echo "SPI end    $SPI_END"
echo "SPI size   $SPI_SIZE"

# build OTGW-firmware-mklittlefs.bin
# logic used from arduino-esp8266littlefs-plugin
# https://github.com/earlephilhower/arduino-esp8266littlefs-plugin
$MKLITTLEFS -c $DATAPATH -p $SPI_PAGE -b $SPI_PAGE -s $SPI_SIZE $IMAGE_LITTLEFS
```

rlagerweij, 2022-01-25:
i'm interested in this process. From the look of it you have a small typo. it should read "-b $SPI_BLOCK"  and not what is there currently, which is "-b $SPI_PAGE"

rlagerweij, 2022-02-06:
additionally, you are using memory mapped flash addresses instead of physical memory locations. i.e. the SPI_START should be 0x200000 instead of 0x40400000. same for SPI_END, so subtract 0x40200000 from that as well

jvwdev, 2022-02-06:
@rlagerweij thank you for posting what you found! Do you have it in a working state? When I receive my order of new boards, I'll give it a shot

rlagerweij, 2022-02-07:
I have the littlefs part working, but I am not working on the arduino-cli part.

mikeg-de, 2022-09-30:
@rlagerweij I struggle to follow your steps to obtain the OTGW-firmware.mklittlefs.bin file. Can you share more details please? Or, @rlagerweij can you add the missing file in the repo instead please?

rvdbreemen, 2022-09-30:
There is really no need to compile the firmware yourself, as I precompile all the binary's for you to use. Just get the the binary releases files here: https://github.com/rvdbreemen/OTGW-firmware/releases

closed 2022-09-30 by rvdbreemen