rvdbreemen/OTGW-firmware #82: Read thermostat setpoints from WRITE_DATA instead of WRITE_ACK as boiler may not support data ID

binsentsu, 2021-12-07:
Hi.

Example: My boiler does not support ID=16.
In this case the control setpoint keeps being 0° as the value is read from the WRITE_ACK line.
Through OTMonitor the correct value is read. Because of these possible unsupported data ids, the native HA integration reads some data points from the WRITE_DATA iso WRITE_ACK message.  (reference for Trset: https://github.com/mvn23/pyotgw/blob/1854ef4ffb907524ff457ba558e4979ba7fabd02/pyotgw/messages.py#L206, https://github.com/mvn23/pyotgw/blob/1854ef4ffb907524ff457ba558e4979ba7fabd02/pyotgw/protocol.py#L228)

Debug log of OTGW-Firmware in case data id is not supported by boiler:
`18:30:03 (  15544| 11960) processOTGW (1225): Thermostat       T10101380 (9)[10101380][MsgID= 16][WRITE_DATA      ][TrSet                         ]	hb[ 19] lb[128]
18:30:03 (  17136| 11960) processOTGW (1221): Boiler           B70100000 (9)[70100000][MsgID= 16][UNKNOWN_DATA_ID ][TrSet                         ]	hb[  0] lb[  0]
18:30:03 (  15792| 11960) processOTGW (1233): Answer Themostat AD0100000 (9)[d0100000][MsgID= 16][WRITE_ACK       ][TrSet                         ]	TrSet = 0.00 °C`

Does this make sense?

rvdbreemen, 2021-12-07:
@binsentsu I am actually in the process of trying to fix this issue you describe here. The OT message decoding has changed. If you join the Discord community you can help test the new firmware release. That should be able to correctly work.

rvdbreemen, 2021-12-20:
@binsentsu the bughunt is over, the OT decoding got fixed. If you update to release v0.9.1 and find you still have issue, then let me know by reopening the issue.

closed 2021-12-20 by rvdbreemen

binsentsu, 2022-01-02:
Hereby confirmed to be fixed in 0.9.1. Thanks!