rvdbreemen/OTGW-firmware #83: Constant gateway reset (msgIDs are missing) due to high Mqtt message load

Richard-ha, 2021-12-08:
Equipment bought from nodoshop:
Nodoshop OTGW + ESP (latest version)

All flashed with latest firmware. It works fine, but lots of msgIDs are missing.
 FYI, Returnwater temperature and remote setpoint override doesn't work
Trying to find the problem with OTmonitor.
Log tab: There is a constant (every 5 to 10sec) gateway reset going on while reading the OT messages

If checking for error code in the OTGW: there is a powerdrop below 4.35 volt
Disabling Mqtt solves the problem. It looks like the huge amount off Mqtt information send (high interval rate) overwhelms the ESP.
API calls are now working correct, all data is received, properly.

Option in settings: set Mqtt interval

rvdbreemen, 2021-12-08:
@Richard-ha not sure what is going on in your situation, you are reporting multiple issues at once, some related to hardware (voltage drop), not reported msgid's (eg. returnwater temperature) or unprocessed messages. 

It seems when you connect using the serieal connections (using USB I assume) then you seem to get it working and with OTmonitor you can see the msgid's and remote override works

You report the API calls are working correctly. But the MQTT is not working for you. 

To troubleshoot this over slowchat is hard, would it be possible to join the discord. There are more people there that can and will help to figure out what is going on.

Also it helps to share a logfile on what you see.

rvdbreemen, 2021-12-08:
Actually, I have seen your remark, but I don't see message loss with the firmware in normal conditions, I wonder what you see as number of messages from OpenTherm itself. Do you have an idea on how many messages you see flowing on port 25235?

Richard-ha, 2021-12-08:
Hi Robert,
Thank you for the quick response. I guess there could also something wrong with the (long >20meters) and maybe bad cable between the boiler and thermostat. To find out I'm going to mount the thermostat in close range from the boiler and perform some new tests. 

Could you please forward me the link to the discord server in case it didn't work.
Tnx in advance

rvdbreemen, 2021-12-08:
The discord can be found here: https://discord.gg/zjW3ju7vGQ

Richard-ha, 2021-12-08:
Sorry cancelled my subscription from discord.
I don't giveaway my phone number to subscribe. Took al verification steps, that must be enough. I will keep communicate via github, sorry for that.

Verzonden via Yahoo Mail op Android

rvdbreemen, 2021-12-08:
Thats fine with me :)

rvdbreemen, 2021-12-08:
I actually allow you to join the discord without phone verification, so it must be a discord thing that they ask for your phone number. I only require members to be online for more than 5 minutes on discord, to prevent bots from spamming us.

binsentsu, 2021-12-08:
Don't know if it helps, but I suffered constant gateway resets in the beginning when the OTGW was powered through a 3m USB cable. I replaced it for a 30cm cable and since then all reset problems were gone.

rvdbreemen, 2021-12-09:
@binsentsu thats a great observation. You need to make sure that the OTGW USB is fed with 5V. If you use long cables and a weak 5V adaptor it will start to reset. I had that myself too and now use a 30cm cable and a 2A adaptor (overkill but it works). 

@tjfsteele any additions on the power situation on V2.3?

Richard-ha, 2021-12-09:
@binsentsu , thanks that did the trick.

New (thicker,  better quality) USB kabel  and 2A Samsung fast USB charger solved all the problems. The original one delivered from nodoshop is 1A. 
Mqtt running perfect, override temperature works and no more gateway resets or missing ID's
Thank you all for the great tips.

rvdbreemen, 2021-12-09:
Closing the ticket.

closed 2021-12-09 by rvdbreemen

tjfsteele, 2021-12-09:
Shouldn't need anything like 2A but a decent voltage at the board is a good
idea. Sounds like the manual could be updated.

Tim

Richard-ha, 2021-12-09:
I agree Tim,
I think indeed it needs a decent stable power supply, 1A should be more than enough. Although if cheap components are used in the delivered charger the riple at higher powerconsumption could be the case. I didn't measure the voltage on the board.
I assume the original Samsung USB charger is capabele delivering stable 5 Volt.
But it is solved, that's more important.Richard