rvdbreemen/OTGW-firmware #84: DWH Enable flag not correct in case thermostat flag overridden through OTGW config

binsentsu, 2021-12-09:
Setup:
- Thermostat always sets DHW_ENABLE =1 within msgID = 0
- Through HW=0 configuration within OTGW, this behaviour is overridden towards the boiler.

Outcome:
- DHW_ENABLE = 1 published through mqtt.

Expected:
- DHW_ENABLE = 0 published through mqtt.

I suppose reason is that the overridden DHW_ENABLE BIT is never read from the transformed message (R00000000) from the OTGW towards the boiler, hence the 'not processed, received from OTGW' message.

OTGW log:
```
23:58:08 (  17184| 15376) processOTGW (1225): Thermostat       T80000200 (9)[80000200][MsgID=  0][READ_DATA       ][Status                        ]	Status = Master [-D---S--] 
23:58:08 (  15816| 14728) sendMQTTData( 301): Sending MQTT: server 192.168.1.151:1883 => TopicId [OTGW/value/otgw-A848FACEAA9B/status_master] --> Message [-D---S--]
23:58:08 (  15168| 14080) sendMQTTData( 301): Sending MQTT: server 192.168.1.151:1883 => TopicId [OTGW/value/otgw-A848FACEAA9B/ch_enable] --> Message [OFF]
23:58:08 (  14496| 13432) sendMQTTData( 301): Sending MQTT: server 192.168.1.151:1883 => TopicId [OTGW/value/otgw-A848FACEAA9B/dhw_enable] --> Message [ON]
23:58:08 (  13800| 12784) sendMQTTData( 301): Sending MQTT: server 192.168.1.151:1883 => TopicId [OTGW/value/otgw-A848FACEAA9B/cooling_enable] --> Message [OFF]
23:58:08 (  13824| 12784) sendMQTTData( 301): Sending MQTT: server 192.168.1.151:1883 => TopicId [OTGW/value/otgw-A848FACEAA9B/otc_active] --> Message [OFF]
23:58:08 (  13824| 12784) sendMQTTData( 301): Sending MQTT: server 192.168.1.151:1883 => TopicId [OTGW/value/otgw-A848FACEAA9B/ch2_enable] --> Message [OFF]
23:58:08 (  15128| 12784) sendMQTTData( 301): Sending MQTT: server 192.168.1.151:1883 => TopicId [OTGW/value/otgw-A848FACEAA9B/summerwintertime] --> Message [OFF]
23:58:08 (  15144| 12784) sendMQTTData( 301): Sending MQTT: server 192.168.1.151:1883 => TopicId [OTGW/value/otgw-A848FACEAA9B/dhw_blocking] --> Message [OFF]
23:58:08 (  15168| 12784) print_status( 566): Status u16 [0200] _value [0200] hb [02] lb [00]
23:58:08 (  15168| 12784) processOTGW (1426): Not processed, received from OTGW => [0000] [4]
23:58:09 (  17184| 15376) processOTGW (1221): Boiler           BC0000000 (9)[c0000000][MsgID=  0][READ_ACK        ][Status                        ]	Status = Slave [--------] 
```

OTMonitor log:
```
23:56:28.971511	T80000200	Read-Data 	Status (MsgID=0): 00000010 00000000
		   - CH enable: disabled (0)
		   - DHW enable: enabled (1)
		   - Cooling enable: disabled (0)
		   - OTC active: not active (0)
		   - CH2 enable: disabled (0)
		   - Summer/winter mode: winter (0)
		   - DHW blocking: unblocked (0)
23:56:29.022648	R00000000	Read-Data 	Status (MsgID=0): 00000000 00000000
		   - CH enable: disabled (0)
		   - DHW enable: disabled (0)
		   - Cooling enable: disabled (0)
		   - OTC active: not active (0)
		   - CH2 enable: disabled (0)
		   - Summer/winter mode: winter (0)
		   - DHW blocking: unblocked (0)
23:56:29.088313	BC0000000	Read-Ack  	Status (MsgID=0): 00000000 00000000
		   - Fault indication: no fault (0)
		   - CH mode: not active (0)
		   - DHW mode: not active (0)
		   - Flame status: flame off (0)
		   - Cooling status: not active (0)
		   - CH2 mode: not active (0)
		   - Diagnostic indication: no diagnostics (0)
		   - Electricity production: not active (0)
23:56:29.170815	A40000200	Read-Ack  	Status (MsgID=0): 00000010 00000000
		   - Fault indication: no fault (0)
		   - CH mode: not active (0)
		   - DHW mode: not active (0)
		   - Flame status: flame off (0)
		   - Cooling status: not active (0)
		   - CH2 mode: not active (0)
		   - Diagnostic indication: no diagnostics (0)
		   - Electricity production: not active (0)
```

rvdbreemen, 2021-12-10:
Hi @binsentsu thanks for finding an analyzing this issue. I published a build of beta 0.9.1 in the discord last night. Did you capture using this firmware?

In other words what version of the firmware did you use to capture this log. Btw. this is an override issue for mqtt, and I was already trying to fix that, but as I did not have this configuration I had nog seen the logs yet myself. So this helps.

If you have not used the build from last night (9/12) then please do and let me know what you find now.

Thanks
Robert

binsentsu, 2021-12-10:
Hi. Didn't check discord last night. Will do this evening. This log is captured through the latest released firmware version: 0.9.0+21bea01

rvdbreemen, 2021-12-20:
@binsentsu I think we solved the issue with the latest release v0.9.1. If not then please reopen this issue with new logs to show me.

closed 2021-12-20 by rvdbreemen