rvdbreemen/OTGW-firmware #85: MQTT messages settings with checkboxes

PvdGulik, 2021-12-10:
Hello Robert,

Still using your nice firmware.
A suggestion what I would like to see is:
A web-page with all the possible messages/devices/settings. Why? A lot of these are not working for a lot of boilers. But connected to Domoticz, it creates a lot of devices which are usuless. That's why it would be nice if you could select which ones you would like to see.

Another suggestion would be you could change the topic.. So default the name you send, but als editable. 

These two suggestions would be nice on one sub-web-page.

Thanks, Pieter

rvdbreemen, 2021-12-10:
Hoi Pieter

Thanks for the great suggestions. Actually the devices are created using the configuration file which can be stripped if you know what you are doing. That way the devices would not appear in Domticz. 

I will put them on the backlog to investigate. 
Groet
Robert

rlagerweij, 2021-12-23:
I have some proof-of-concept code that only sends the MQTT configuration topics of messages it actually encounters. This should cut down on the useless enities in Home Assistant. The POC works fine for regular sensors but not currently for binary sensors since there is one message which needs to trigger a lot of different topics. 

One solution could be to rename some of the topics to include the label of the related MGGid.
e.g rename the "fault" topic to " Status_fault" and the "service_request" to " ASFflags_service_request" 

I'm not quite sure what other consequences there may be from such renaming since I have not really dug into the MQTT side of things.

rvdbreemen, 2021-12-23:
I am interested... So you send the entities on the fly as they occur for the first time? How do you match them up with the actual message id? I assume you go thru the configuration file very time you see a message, or something similar. Do you keep a list of configured topics? Like I said, you got my attention on this one.

I would like to suggest instead of renaming all topics, we just extend the file format of the mqttha.cfg file, by simply adding one value in front of it...

Like this:
<msgid>,<configtopic>,<content op config>

By extending it by one, you make it easy go thru the configuration file and just transmit all lines that match the message id. You don't have to look for string matches, and thus you don't have to change the topic names.

rlagerweij, 2021-12-23:
> I am interested... So you send the entities on the fly as they occur for the first time? How do you match them up with the actual message id? I assume you go thru the configuration file very time [...]
That is exactly what I am doing indeed

> I would like to suggest instead of renaming all topics, we just extend the file format of the mqttha.cfg file, by simply adding one value in front of it...
>
> Like this: [...]

Sounds like a good idea. I'll try this approach

rvdbreemen, 2021-12-23:
@rlagerweij actually I just realized that you don't need to keep your own list, you could simply check the ```msglastupdated[xxx]``` to see if it is still zero, because then it means it's the first time you see that msgid.

rlagerweij, 2021-12-23:
I didn't know that. that saves another 24 bytes of memory.

rvdbreemen, 2021-12-23:
@rlagerweij if we are going to do this. Then I suggest we move the file into flash memory, that way we can have a static lookup tabel and we don't have to parse the config file each message.

rlagerweij, 2021-12-24:
we could do that, but the file only gets parsed maybe 20 times in the first few minutes after startup so I don't think it is technically necessary for performance reasons. 

Would you object to changing the delimiting character from "," to something like " ;".  Since there are a lot of comma's in the json in each line. Probably also not technically necessary but i feels a bit safer.

rvdbreemen, 2021-12-24:
It's not about performance, but more about the simple fact that you can more easily walk a array of strings by index, then you can walk a file. With a in-flash array you don't need to parse anymore. 

No I don't mind the delimiter, even though it should not matter at this point. As the first two comma's will never have comma's in their strings, only the last one. You may have noticed I only split on the first ',' at this point.

rvdbreemen, 2022-01-08:
This selective mqtt discovery feature was merged and released in v0.9.2. To remove unwanted topics from configuration, you can change mqttha.cfg

closed 2022-01-08 by rvdbreemen