rvdbreemen/OTGW-firmware #88: Can't change remote setpoint

rvoosterhout, 2021-12-13:
Hi Robert,

In regards to your PM, I'm posting the same question I also posted on the domoticaforum.eu here.

For original post: [https://www.domoticaforum.eu/viewtopic.php?f=75&t=13381&sid=132eba8a6c9ba8285d08bb4a1444b2f9](url)

I started to do some investigating, and I got it mostly to work. When I remove the D1 mini, connect the otgw to my PC with otmonitor and I connect the Remeha iSense base station, it's working. I get a lot of data from my boiler and I can control the remote override room setpoint using the Thermostat menu (F2). Because I'm using a Remeha iSense RF, it does take 12 minutes for the override setpoint to reach the thermostat, but in the end, it does work.

Now when I connect the D1 mini to the OTGW, I found 2 problems.

- The Remeha thermostat doesn't connect anymore to the base station
I found that the D1 mini uses too much power, for the base station to work. I'm using a 5V 2A USB connector, but this is not sufficient. Luckily, the iSense base station has an additional 5V input, so I connected a 5V 4A power supply to that and now the thermostat can connect to the base station again.

- I can't control the remove override room setpoint anymore.
When I'm using otmonitor on my laptop, with the D1 mini connected to the otgw, setting the remove setpoint (F2) doesn't send this value to the thermostat. In the log, I do see the Command: TT=25, and after that the TT = 25 confirmation message, but after 12 minutes, when the iSense thermostat should change, nothing happens.
I also turned on MQTT again, and tried to set the setpoint using a thermostat card in homeassistant, same story. I do see the Command: TT=25 (or the value I set on the thermostat in homeassistant) in the logging, but after that, no confirmation, which I do get when setting the setpoint using otmonitor.

rvdbreemen, 2021-12-13:
Hi Rick,
Thanks for logging your issues here, you can find me online most of the time on Discord [(link)](https://discord.gg/zjW3ju7vGQ).

In response to your issue:
1. Powerissue
Does the iSense base station take it's powerfeed from the 24V to generate the power needed? I ask this because the V2.3 hardware uses the 5V to boost to 24V. When the power need of the basestation of the iSense stays within specification it should work, but sure sounds it's an issue in your scenario. Maybe @tjfsteele (hardware designer) will be interested in this particular scenario. Glad to hear you figured out this particular issue and fixed it. Thanks for reporting that here too.

2. Cannot override room setpoint. 
It sure sounds something I need to try to fix. For that I would need your help. Best would be if you join us on Discord, there I work with community members to solve these type of issues. To help me you should share some logs of both OTmonitor and the telnet port 23 (debuglogging of firmware). That way I can try to understand what is going on and fix it.

Hope you join us in an effort to fixing this, as a lot of people use iSense thermostats, so if there is a bug, then we need to fix it.

Robert

tjfsteele, 2021-12-13:
The 24V power supply on the OTGW does come from the 5V rail. From what you
are saying it seems the 5V rail is not sufficient to power the WeMos as
well as the OTGW and attached devices which is very surprising.

I am certainly interested to find out more!

Tim

rvoosterhout, 2021-12-13:
Hi Tim,

Please let me know what I can check to investigate this. I'm using a Samsung travel adapter for the OTGW board psu. It's a 5v 2a unit.
The iSense basestation is now connected to a 5v 4a powersupply, not USB so I can't use that one for the otgw board.

tjfsteele, 2021-12-13:
Ideally we would like to know how much current the WeMos is consuming.

Thanks

Tim

Roos-AID, 2021-12-13:
Both iSense and WeMos are consuming significant more during Wifi setup/connection than I ever expected.  I don't know the exact amount of current required, but from experiments with ESP8266 Lolin and NodeMCU I know that it went beyond 250 mA.  Add to that the display in the iSense etc. and I think that the 5V to 24V generation is causing troubles with the 5V rail stability required for the WeMos.  Regards, Rob

rvoosterhout, 2021-12-13:
To clarify, i'm using the iSense RF. My thermostat is wireless. There are 4 AA batteries in the thermostat to power the thermostat itself. The base station is connected to the otgw.

rvdbreemen, 2021-12-13:
Have you measured the power use of the base station? I know you solved it by putting the base station on it's on supply, but I wonder what is really going on powerwise.

eXistenZNL, 2021-12-15:
Joining this topic because I have the exact same issue but with different hardware:

Boiler: [Itho Daalderop Aqua-Max](https://www.ithodaalderop.nl/nl-NL/professional/product/545-1642)
Thermostat: [Honeywell Chronotherm Vision Modulation](https://shop.kamerthermostaat.com/p/honeywell-chronotherm-vision-modulation)

- When connected directly to my laptop, without the WeMos D1 Mini installed, all works well.
- When the WeMos D1 Mini is installed, using a 1A USB power supply, the OTGW restarts every second*
- When the WeMos D1 Mini is installed, using a 2A USB power supplly, the OTGW restarts every 5-10 seconds*

* I think it's power cycling because leds A-E blink together instead of showing the status normally.

When the OTGW restarts that often, it is impossibe for me to send data, only to receive it.

tjfsteele, 2021-12-15:
Have you tried a shorter USB cable?

eXistenZNL, 2021-12-15:
@tjfsteele thanks for responding so quickly!

Good idea, haven't tried that yet, as USB-B is quite hard to come by in my household.

Just ordered [this cable](https://www.allekabels.nl/usb-printer-kabel/4578/1285874/printerkabel-usb-a-naar-usb-b.html) which is only 25cm long instead of the current 2 meter cable I have, and will post the results tomorrow after testing.

tjfsteele, 2021-12-15:
That should help. Also a good idea to use a 28+24awg cable if you can.

Tim

eXistenZNL, 2021-12-16:
@tjfsteele we have succes!

The OTGW seems to have enough power only with the combination of a 25cm cable and a 2A adapter. Even when using the 1A adapter with the 25cm cable, the OTGW still keeps rebooting when the WeMos is plugged in.

Even though I solved the issue, it seems that a few devices out there rely rather heavy on pulling power via the 24V line.
I feel that v2.3 of this board only delivers the bare minimum in that regard and suppose that if I tried one of the original / older designs with a dedicated 12v or 24v adapter, I might not have had this problem at all.

Would it be a wise idea to add an optional 12V rail for some more grunt on future revisions of this board?

Anyway, thanks for helping me solve this by suggesting a shorter cable.

tjfsteele, 2021-12-16:
I appreciate your feedback and we will consider improvements in future
versions

Best wishes

Tim

rvdbreemen, 2021-12-20:
@rvoosterhout do you mind if we close this issue. I think we solved it, and @tjfsteele even got feedback for a future release of the hardware.

rvoosterhout, 2021-12-21:
Sure, no problem. I'm going to return my otgw to nodoshop for investigation.

rvdbreemen, 2021-12-21:
@rvoosterhout good to hear that you are in contact with nodoshop. Looking forward to having you back when you have your working OTGW from nodosop.

closed 2021-12-21 by rvdbreemen

erwinwildenburg, 2021-12-29:
Hardware:
Remeha Calenta Ace 40c
Evohome (R8810A1018)
Nodo OTGW 2.3 with USR (presolderd)

I think I have the same problem. Using a Samsung travel charger 5v/2A with a short 0.25m cable solved the power issue partly.

When using it with only USB power it works fine. When plugging in the network cable after a minute or so otmonitor gives the message: "High power usage!" and then the thermostat disconnects.

It feels like there is a bigger issue here.

rvdbreemen, 2021-12-29:
@erwinwildenburg it sounds a bit different to me to be honest. As you use a networkcable, the issue with the USB power and cable has been seen only in combination with wemos. Maybe @tjfsteele could comment on what could be going on with "high power usage". 

As you see this message in otmonitor, you could also ask Schelte Bron on his forum here: https://www.domoticaforum.eu/viewforum.php?f=75

erwinwildenburg, 2022-01-08:
Hi,

I had some time to look at this again.
It turns out updating the firmware to 5.2 fixed the problem for me.

The changelog also mentions a lot about voltage:
https://otgw.tclcode.com/changelog.html

rvdbreemen, 2022-01-08:
@erwinwildenburg that's great news that the firmware update fixed your issue. Then there must have been a smartpower issue in your case, because that is the core of the change on the 5.2 firmware if I am not mistaking.

Thanks for letting me know.

gregiOS, 2022-10-01:
Hello @rvoosterhout did you manage to resolve the issue with not being able to set room setpoint.
I have the exactly the same issue with my iSense thermostat and DeDritch heater.