rvdbreemen/OTGW-firmware #89: OTGW randomly stops

PuckeyP, 2021-12-15:
Hi,
I'm having a problem with the OTGW - NodeMCU in combination with Remeha Calenta and Remeha iSense.
I'm using NodeRed /Home assistant to control the remote override setpoint.

All has worked well for about three months, but suddenly the OTGW randomly stops.
Connection to the OTMonitor is lost, logging stops, and ch temperature is set to 14 degrees.
Any ideas where to look for an error
Thx,
PPee

tjfsteele, 2021-12-15:
Can you confirm which version of OTGW you have?

Thank you

>

PuckeyP, 2021-12-15:
Hi tjfsteel,
Details from the webui - Device Info:
- NodeMCU Firmware Version 0.9.0+21bea01 (06-11-2021)
- OTGW PIC Firmware Version 5.1
- Compiled on (date/time) Nov 6 2021 16:09:58
- Arduino Core Version 2_7_4
- Espressif SDK Version 2.2.2-dev(38a443e)

tjfsteele, 2021-12-15:
Can you tell me the hardware version? Should be written on the OTGW board
e.g. 2.0

>

rvdbreemen, 2021-12-15:
@PuckeyP could you share some logging from your OTGW firmware, just telnet to port 23 and capture some logs and share them here. Maybe we can see if anything weird is going on.

PuckeyP, 2021-12-16:
I attached the complete TelNet log. At 12:00 logging stopped unexpectedly. At 11:43:42 ch_enabled suddenly turned to "OFF"
Hope you can find a indication of the error.

PuckeyP, 2021-12-16:
[putty.log](https://github.com/rvdbreemen/OTGW-firmware/files/7727206/putty.log)
File attached

PuckeyP, 2021-12-16:
Hi Robert that is strange, because I did not use hot water this morning. Nobody was home until 13:00...

rvdbreemen, 2021-12-16:
Hi @PuckeyP well, then that's very weird indeed, but that's what your boiler is reporting back to the OTGW according to your logs. Does your boiler have an error log history?

rvdbreemen, 2021-12-20:
@PuckeyP just release version 0.9.1 that fixes the OT decoding. So please try that. I will close this issue in the mean time, if you still have problems, then please reopen this issue.

closed 2021-12-20 by rvdbreemen