rvdbreemen/OTGW-firmware #90: Option: Enable PIC compatibel messages on serial interface.

STemplar, 2021-12-16:
Perhaps it is a good idea to have Opentherm Monitor compatability and have OTGW style serial reporting without parsing the Data-ID's. In other words, an option to enable legacy OTGW reporting over Net2Serial.
[image]

rvdbreemen, 2021-12-16:
@STemplar what do you mean? I don’t understand your request. Please elaborate.

rvdbreemen, 2021-12-16:
If I understand your request correctly, then I think you looking for option 3) serial2net. 

That is ‘legacy’ support. It was the first thing my firmware did, and I continue to support it. You can use OTmonitor (from Schelte Bron) or any other serial over network based integration. 

Please confirm if that is what you mean? And if not, then please do elaborate your request. 

Robert

STemplar, 2021-12-19:
Ah, that is it! I needed to use port 25238! You can close this Issue.

closed 2021-12-19 by STemplar