rvdbreemen/OTGW-firmware #98: Feature request: Disable WiFi traffic LED

fsfikke, 2021-12-22:
Hi,

I currently use the OTGW with a NodeMCU (ESP8266_NODEMCU), this has a bright blue Builitin LED on GPIO02:
[image]
https://lowvoltage.github.io/2017/07/09/Onboard-LEDs-NodeMCU-Got-Two

This lights up the whole room where my boiler is located at night. It would be great if this could be turned off by a software setting.

rvdbreemen, 2022-01-02:
Thanks for reporting it. The setting in the Webui should turn of both blue LEDS. But it was broken when we moved from one to two led signaling. Thus I fixed this in the dev release currently under way, it will be released as part of 0.9.2.

rvdbreemen, 2022-01-08:
fixed in release v0.9.2

closed 2022-01-08 by rvdbreemen