rvdbreemen/OTGW-firmware #100: Cannot change wifi settings after initial setup

drstikko, 2021-12-27:
After connecting to the device (mine is Wemos D1 mini pro) in AP mode after flashing, one selects the SSID/password for local network connection. Unfortunately it seems it cannot be changed afterwards (maybe I overlooked something). This can be unhandy when for example the wifi password of the router changes.

DaveDavenport, 2021-12-27:
I think if your wifi password change and you reboot (reset) the wemos it should go into the original ap mode.
(I agree a option in the UI would be nice.)

drstikko, 2021-12-27:
Actually, in my case, I want to move it to another access point. So now, it keeps connecting to the 'old' one. I thought actually going to the setup link could work? But I don't have it in my history.

rvdbreemen, 2021-12-27:
The easiest way to re-configure the wifi is to simply turnoff or change the password on the wifi accesspoint it is connecting too. 

That way it will reconfigure enter the configuration mode once more. 

It's on my longlist to improve this. But this is the work around for now.

DaveDavenport, 2021-12-27:
@rvdbreemen  maybe we can add a button that calls:

```C++
WiFiManager::resetSettings();
```
or even just starts the portal?

rvdbreemen, 2021-12-27:
Yep, that's the idea. Just reset the settings...

drstikko, 2021-12-27:
sounds good

DaveDavenport, 2021-12-27:
Added an PR: #101

rvdbreemen, 2021-12-31:
Will include reset button in next release.

rvdbreemen, 2022-01-08:
Fixed in release 0.9.2

closed 2022-01-08 by rvdbreemen