rvdbreemen/OTGW-firmware #106: Toon: Intermintting error in communication to boiler

SanderV4n, 2022-01-01:
Hi!

I got a version 2.3 OTGW installed between my boiler (Remeha Calenta) and the Toon boiler communication module (ketelmodule). With the latest version (17) I get a pop-up on Toon  that there is no communication with the boiler. These errors seem to last from a few seconds to a few minutes.

On the previous version (16) I did not see these pop-ups, but also did not run it that long as the new release became available.

With the error active, changes in setpoint from Toon to the boiler don't seem to get applied.

Any thoughts?

rvdbreemen, 2022-01-01:
@SanderV4n please provide some logs using these instructions:
https://github.com/rvdbreemen/OTGW-firmware/wiki/How-to-debug-the-OTGW-firmware

Share the logs. Or join us on discord to discuss the findings.

SanderV4n, 2022-01-02:
Hi @rvdbreemen 

Here is the log file attached:
[otgwdump.txt](https://github.com/rvdbreemen/OTGW-firmware/files/7798972/otgwdump.txt)

rvdbreemen, 2022-01-02:
@SanderV4n from your log it looks like your OTGW is constantly resetting. You can easily recognize this by seeing the version banner of the PIC coming back all the time. This is the banner: "OpenTherm Gateway 5.1"

This is a result of brownouts on the PCB. You can try using a short high quality USB cable, and a high power adapter (1A+). If this does not work, you need to contact Nodoshop.

closed 2022-01-02 by rvdbreemen