rvdbreemen/OTGW-firmware #108: DWH Enable flag flipping in case thermostat flag overridden through OTGW config

binsentsu, 2022-01-02:
Follow up of issue #84 with OTGW firmware 0.9.1
The DHW enabled bit is now read, but is constantly flipping between 1 and 0.
Use case is still the same.
Setup:

- Thermostat always sets DHW_ENABLE =1 within msgID = 0

-  Through HW=0 configuration within OTGW, this behaviour is overridden towards the boiler.

Reason is that the DWH Enable = 1 is read from the T-message, directly published to mqtt. Thereafter OTGW does an override through the R-message to 0, then this message is sent through mqtt. And so on...

```
15:19:10.180835	T80000200	Read-Data 	Status (MsgID=0): 00000010 00000000
		   - CH enable: disabled (0)
		   - DHW enable: enabled (1)
		   - Cooling enable: disabled (0)
		   - OTC active: not active (0)
		   - CH2 enable: disabled (0)
		   - Summer/winter mode: winter (0)
		   - DHW blocking: unblocked (0)
15:19:10.289333	R00000000	Read-Data 	Status (MsgID=0): 00000000 00000000
		   - CH enable: disabled (0)
		   - DHW enable: disabled (0)
		   - Cooling enable: disabled (0)
		   - OTC active: not active (0)
		   - CH2 enable: disabled (0)
		   - Summer/winter mode: winter (0)
		   - DHW blocking: unblocked (0)
15:19:10.570528	BC0000000	Read-Ack  	Status (MsgID=0): 00000000 00000000
		   - Fault indication: no fault (0)
		   - CH mode: not active (0)
		   - DHW mode: not active (0)
		   - Flame status: flame off (0)
		   - Cooling status: not active (0)
		   - CH2 mode: not active (0)
		   - Diagnostic indication: no diagnostics (0)
		   - Electricity production: not active (0)
15:19:10.681877	A40000200	Read-Ack  	Status (MsgID=0): 00000010 00000000
		   - Fault indication: no fault (0)
		   - CH mode: not active (0)
		   - DHW mode: not active (0)
		   - Flame status: flame off (0)
		   - Cooling status: not active (0)
		   - CH2 mode: not active (0)
		   - Diagnostic indication: no diagnostics (0)
		   - Electricity production: not active (0)
```
[image]

rvdbreemen, 2022-01-02:
@binsentsu did you try the beta 0.9.2-beta yet, because I think that should solve this behavior. Could you verify that for me?

rvdbreemen, 2022-01-02:
Or wait, is this a new usecase?
Again, could you also provide the logs when using beta 0.9.2 and OTmonitor, just to see if they match now?

rvdbreemen, 2022-01-02:
Just went and checked your use-case... I stand corrected, it's a bug in the status sending. 
So I will fix it now right away, and update the current beta. 
If you could validate that, then I will fix this asap.

rvdbreemen, 2022-01-08:
fixed in release 0.9.2

closed 2022-01-08 by rvdbreemen