rvdbreemen/OTGW-firmware #131: Are there any way to block the public wifi when router is offline?

rotilho, 2022-07-27:
I had to disconnected my router for several hours few days ago and I noticed the gateway starts the public wifi after some time but reconnects if my router comes back online.

Are there any way to block that? I don't want to expose my boiler if my router is offline so it would great if the gateway stick to the configured wifi.

rvdbreemen, 2022-08-18:
Well. The whole setup was made so with intermittent WiFi it just recovers. No open WiFi. But after an extended period it will assume it needs to be reconfigured that's why you see the WiFi auto configuration portal pop up. 

If you restore wifi it should recover without an issue, even when it is already in configuration mode. 

Not sure how to improve this behavior to be honest without losing functionality. Any suggestions?

rotilho, 2022-08-18:
I understand that for first install it totally makes sense but after it's already configured I'd expect that I would have to reset it to pair to a new network. This is similar behavior to ESPHome (when the wifi credentials are not compiled together) and most of Zigbee devices.

I just want to avoid that technical issues with my router cascade to my boiler being exposed.

rvdbreemen, 2022-08-18:
So you are suggesting that you should reset after a certain amount of time to recover when an WiFi PSK or SSID changes?

It that the fix your are proposing?

rotilho, 2022-08-21:
No no no, sorry. What I'm proposing is:

- If there's no wifi configured: gateway create a public wifi so we can configure it
- If there's an wifi configured and it's not connected: keep retrying every X time
- If there's an wifi configured and I want to change the wifi: I need to manually reset the gateway with the physical reset button in the gateway

dwar, 2022-11-26:
I think this is a security vulnerability! 
The wifi credentials are stored on esp.
- Jam the wifi until the recovery AP should be up. (in my case random dropouts :-( )
- Connect to the AP before the esp reconnects to the original wifi. 
- Upload new firmware which exposes the wifi password. 
- Your in

DutchEngineer, 2022-12-16:
Would it be an option to always have a password in AP mode? Having a default password after flashing the firmware. And having the option to change the password in the settings. This way we would keep all the features while still having some security?

0crap, 2022-12-20:
> I think this is a security vulnerability! The wifi credentials are stored on esp.
>
> * Jam the wifi until the recovery AP should be up. (in my case random dropouts :-( )
> * Connect to the AP before the [...]

If this is true then this issue needs urgent attention!
The OTGW uses standard telnet and a webGUI without any authentication, which is already not the greatest idea.
However, if it's possible to extract the WiFi password in whatever way, that is a serious vulnerability that needs patching.

For the initial setup it's fine as it is, but after it's connected to the WiFi network it should only retry on disconnect.
If you need to change the network or WiFi password, just flash the WeMos D1 to the initial setup config.
Could that be a simple solution to quickly mitigate this issue?

There is no excuse to leave this vulnerability in.
A large amount of users will never change the WiFi password, that's why it must stay safe at all cost.
(Just stay on the old OTGW firmware if you don't care about security.)

Hope this gets fixed soon. 👍

rotilho, 2022-12-28:
> That's not a great idea for those of us who can't easily access our OTGW. Tim
> […](#)

Tim, I understand your perspective. While it is true that improving security often involves making some sacrifices in terms of user experience, I believe that the default setting should always prioritize security. If users are comfortable with the less secure option, they can could opt in to it.

rvdbreemen, 2023-01-28:
In v0.10.0 now can no longer just "flash" the firmware from the Hotspot mode anymore. I think this improves security a little, but not enough yet. However, I am not sure if I want to make it more difficult to reconfigure the wifi for users. 

You are welcome to do a PR to improve the security more.

closed 2023-01-28 by rvdbreemen