rvdbreemen/OTGW-firmware #133: Support for 5.4

PvdGulik, 2022-08-03:
Firmware 5.4 is out, does your firmware support the latest firmware?

Version 6.1 (PIC16F1847) / 5.4 (PIC16F88), released 31 May 2022

Fix restarts when no alternatives are available in stand-alone mode.
Don't go into fail safe mode after a WDT reset if no thermostat is connected.
Add Watch dog time-out (W) as a reset reason to the PR=Q command output.
PIC16F1847: Fix detection of thermostat connect/disconnect.
PIC16F1847: Fix random SE responses from AA, DA, MM, PM, and VS commands.
PIC16F1847: Extend debug pointer to addresses above 0xFF.

thanks!

rvdbreemen, 2022-08-18:
Yes, actually I worked on that a while ago. But could not get it to work properly. I need to revisit this. Thanks for the gently reminder.

rvdbreemen, 2023-01-28:
@PvdGulik release v0.10.0 now supports pic16f1846... enjoy the updating!

closed 2023-01-28 by rvdbreemen