rvdbreemen/OTGW-firmware #135: esptool.exe  error: argument --chip/-c: invalid choice: 'p'

ministryofsillywalks, 2022-09-19:
I just tried to flash my d1 mini as described in the wiki.
When running the command as described:
`esptool -cp COM6 -cd nodemcu -cb 921600 -cf ino.bin -ca 0x200000 -cf fs.bin`

I get the following error:
```
usage: esptool [-h] [--chip {auto,esp8266,esp32,esp32s2,esp32s3beta2,esp32s3,esp32c3,esp32c6beta,esp32h2beta1,esp32h2beta2,esp32c2,esp32c6}] [--port PORT] [--baud BAUD] [--before {default_reset,usb_reset,no_reset,no_reset_no_sync}]
               [--after {hard_reset,soft_reset,no_reset,no_reset_stub}] [--no-stub] [--trace] [--override-vddsdio [{1.8V,1.9V,OFF}]] [--connect-attempts CONNECT_ATTEMPTS]
               {load_ram,dump_mem,read_mem,write_mem,write_flash,run,image_info,make_image,elf2image,read_mac,chip_id,flash_id,read_flash_status,write_flash_status,read_flash,verify_flash,erase_flash,erase_region,merge_bin,get_security_info,version}
               ...
esptool: error: argument --chip/-c: invalid choice: 'p' (choose from 'auto', 'esp8266', 'esp32', 'esp32s2', 'esp32s3beta2', 'esp32s3', 'esp32c3', 'esp32c6beta', 'esp32h2beta1', 'esp32h2beta2', 'esp32c2', 'esp32c6')
```

I'm using the latest version of esptool (4.3)

summerfind, 2023-01-14:
how's going there?
any progress?
Thanks

rvdbreemen, 2023-01-14:
@summerfind the wiki has been updated to fix some issues with the latest ESP tool. 

Just follow the instructions here: 
https://github.com/rvdbreemen/OTGW-firmware/wiki/How-to-install-firmware-on-your-ESP

0crap, 2023-01-14:
@rvdbreemen Any particular reason the wiki still suggests to use a tool that is not maintained any more since [2018](https://github.com/igrr/esptool-ck/releases/tag/0.4.13)?

It seems a better solution to move to the [official ESP tools](https://www.espressif.com/en/support/download/other-tools) and get that supported.

rvdbreemen, 2023-01-14:
@0crap nothing wrong with a working tool, even if it's old ;-)

But the official tools will not flash 0.9.5 bin's for some reason. I have not tried to get it to work with esptool.py.

The next release will be flashable with the official ESP tools once more (already verified).

closed 2023-01-14 by rvdbreemen