rvdbreemen/OTGW-firmware #139: White screen on Wifi Configure

0000vt, 2022-09-29:
On an existing setup which used to have the original otgw firmware I'm now flashing OTGW-firmware.0.9.5+4eb7d24.ino.bin on my NodeMCU. 

However, when i press 'wifi configure' or 'info' on 192.168.4.1 I merely get an empty white page. Upgrade page and exit is working when I press it.

Looking forward to any solution

closed 2022-09-30 by 0000vt

GunnarMorrigan, 2025-01-08:
I also ran into this issue. What is the wiki article that helped?

0000vt, 2025-01-08:
Not sure, don’t remember. Make sure you follow all steps one by one.

GunnarMorrigan, 2025-01-08:
I restarted the Wemos d1 mini via the reset button and revisited the page. This time it loaded the wifi page without being blank!

Thanks for the fast response @0000vt