rvdbreemen/OTGW-firmware #141: Compatible with original PCB at https://otgw.tclcode.com/ ?

melkati, 2022-10-09:
Hi.

I have an old PCB as described at https://otgw.tclcode.com/ (I bought from Node Shop few years ago).

Right now I'm using a remote connection with otmonitor an a ESP8266 with Ser2Net (ESP Easy, really).

Can I use this firmware with it? Any modifications to do? 

Thank you!

rvdbreemen, 2022-10-09:
Hi

Yes, if you have an ESP8266 nodemcu version then it should work just fine. It was the main reason why I ever started this project myself. 

So please just try.  And if you cannot get it to work, just visit the discord community and get help. 

Regards
Robert

closed 2022-10-09 by rvdbreemen

melkati, 2022-10-10:
Hello.

I flashed the firmware OTA and it's working without changes!

Problem is... I can't send any commands...

If I try by MQTT they are ignored.

If I try "http://otgw.local/api/v1/otgw/command/TT=20" I receive:

```
OTGW firmware

[/api/v1/otgw/command/TT=20] is not a valid
```

I can do http://otgw.local/api/v0/devinfo and I receive:

```
{"devinfo":[
  {"name": "author", "value": "Robert van den Breemen"},
  {"name": "fwversion", "value": "0.9.5+4eb7d24 (30-05-2022)"},
  {"name": "picfwversion", "value": ""},
  {"name": "picdeviceid", "value": ""},
  {"name": "compiled", "value": "May 30 2022 08:35:36"},
  {"name": "hostname", "value": "OTGW"},
  {"name": "ipaddress", "value": "192.168.1.192"},
  {"name": "macaddress", "value": "84:0D:8E:B1:1A:88"},
  {"name": "freeheap", "value": 13416},
  {"name": "maxfreeblock", "value": 11584},
  {"name": "chipid", "value": "b11a88"},
  {"name": "coreversion", "value": "2_7_4"},
  {"name": "sdkversion", "value": "2.2.2-dev(38a443e)"},
  {"name": "cpufreq", "value": 160},
  {"name": "sketchsize", "value": 553.906},
  {"name": "freesketchspace", "value": 1492.000},
  {"name": "flashchipid", "value": "00164020"},
  {"name": "flashchipsize", "value": 4.000},
  {"name": "flashchiprealsize", "value": 4.000},
  {"name": "LittleFSsize", "value": 2.000},
  {"name": "flashchipspeed", "value": 40.000},
  {"name": "flashchipmode", "value": "DIO"},
  {"name": "boardtype", "value": "ESP8266_NODEMCU"},
  {"name": "ssid", "value": "CALAMAR"},
  {"name": "wifirssi", "value": -75},
  {"name": "ntpenable", "value": "true"},
  {"name": "ntptimezone", "value": "Europe/Amsterdam"},
  {"name": "uptime", "value": "0(d)-00:44(H:m)"},
  {"name": "lastreset", "value": "Software/System restart"},
  {"name": "bootcount", "value": 1},
  {"name": "mqttconnected", "value": "true"},
  {"name": "thermostatconnected", "value": "true"},
  {"name": "boilerconnected", "value": "true"},
  {"name": "gatewaymode", "value": "true"},
  {"name": "otgwconnected", "value": "true"}
]}
```

[image]

melkati, 2022-10-10:
I'll continue on Discord.