rvdbreemen/OTGW-firmware #143: separation of values from thermostat and boiler

antst, 2022-10-22:
I kind of understand that currently MQTT is made to be as compatible as it possible with otmonitor. But, as result, it has the same fundamental issue as otmonitor has (and issue which home assistant integration does not have).

There is no separation for values coming from boiler and thermostat.

For example, there is boiler setpoint control. There are two values coming, one from thermostat (what it wants this set point to be) and one from boiler (what is actual setpoint on boiler). If you override this set point (via topic `ctrlsetpt`), then, effectively, you have two different values, for one topic (`TSet`).
Proper way is to separate values coming from thermostat and from boiler to different topics.

emkookmer, 2022-10-23:
I'm encountering similar problems, also i haven't found a way to detect who is in control (ie. The thermostat of otgw) separating these values  as Antst is suggesting wil help with this

rvdbreemen, 2023-01-28:
@emkookmer if you override the Tset value, the thermostat is no longer in control. It's being overridden by the Gateway. 
Could you share some logs @antst and @emkookmer to show me what you mean. 

Btw what we can do is detect who is controlling the boiler, the gateway or the thermostat. That one is quite easy to add in a future release. Let's do some discussion. If you have joined the Discord community you can talk to me and Schelte on this topic.

rvdbreemen, 2023-01-28:
Love to keep improving the project, so feedback is a must.

hvxl, 2023-10-25:
> Actually, otmonitor has the same issue. (which it didn't have 4 years ago, BTW)

@antst You have reported this multiple times in bug reports against otmonitor as well, but then never respond to questions for more details. I have tested your claims and cannot reproduce them. So please stop spreading rumors about alleged bugs in otmonitor unless you can actually show some evidence.

rvdbreemen, 2023-10-25:
@antst please share logs to show what you mean. It is important to capture the message from port 24 on the ESP. Make sure the OTmessage logging is turned on. 

You may want to turn on the MQTT logging too that way we can see what you mean. 

Like I said before I think there is only one true value for Tset. 

But I love to see if we can have logs to show what you mean. 

Kinds regards
Robert

rvdbreemen, 2026-02-28:
@antst @emkookmer feature has been added to version 1.2.0. Will be released soon. Not sure if this is what you meant back then. But if it is, let me know. 

Btw is you want to test it you can join the discord and ask for the beta.