rvdbreemen/OTGW-firmware #145: Unexpected spikes in MQTT values Water pressure and Relative Modulation level

chokays, 2022-11-05:
[image]
[image]
[image]

There are some spikes in the MQTT data shown in Home Assistant for the water pressure and relative modulation level.

OTMonitor does not show these spikes.

rvdbreemen, 2022-12-20:
You did also report with logging. That helps understanding, but not explaining what is going on. Would you be able to log in full details (with MQTT logging turned on, goto firmware with telnet, and then hit '3'). And see if we can catch the issue in plain sight.

chokays, 2023-01-26:
Duplicate and also fixed

closed 2023-01-26 by chokays