rvdbreemen/OTGW-firmware #146: Comparison with diyless hardware, firmware, library and other alternatives

JeroenvdV, 2022-11-15:
Hi,

I think it would be useful to expand the README.md and add some info comparing this firmware and its capabilities to alternatives. So far I have found the following: 

A library to interface with opentherm https://github.com/ihormelnyk/opentherm_library
An Arduino-adapter for opentherm, seems simpler than the OTGW: http://ihormelnyk.com/opentherm_adapter
An Arduino-sketch to use an ESP OpenTherm Shield with Home Assistant, and other variants: https://github.com/diyless/home-assistant-opentherm-thermostat

These are all parts of the same group of projects by Ihor Melnyk, it seems. As far as I can currently tell, you can use both 'stacks' to do some of the same things:

- Read from and control an opentherm boiler and/or thermostat using Home Assistant and others, or a more low-level API
- Connect via Wi-Fi

However, there are also many differences I can see at first glance:

- Availability of pre-assembled hardware
- Amount and cost of total electronic components of the OTGW vs. the OpenTherm Shield
- Software features of this firmware vs. @diyless variants
- Connection via Ethernet -- There are ESP32 boards that have Ethernet, does the diyless sketch run on those with minor adjustments? Does this repository run on those?
- Connection via USB
- Support for OTmonitor
- Extensive public documentation of boiler and thermostat support

I'm not familiar with the history of each project and perhaps the influence of one on the other. As a prospective user I'd like some help understanding the differences, which is why I propose to add a comparison (by the author) to the README.

JeroenvdV, 2022-11-15:
I also have a few open questions about the functionality of each system and how they could even work together.

- This OTGW-firmware seems to be very full-featured. What software changes would be needed for this to work in conjunction with http://ihormelnyk.com/opentherm_adapter in place of the OTGW device and do away with the OTGW hardware entirely?
- If I understand correctly, this firmware is taking over tasks that OTGW can also do itself. If so, which aspects are still the responsibility of the OTGW PIC firmware?

DaniilMir, 2022-11-23:
> I also have a few open questions about the functionality of each system and how they could even work together.
>
> * This OTGW-firmware seems to be very full-featured. What software changes would be [...]

I just have the same question. 
I was playing around with Ihor Melnik's project and was thinking how to add there a webserial so I could integrate it with hasssio.
It the end I found this project and did a Frankenstein.  First NodeMCU is running opentherm adpater in a gateway mode and a second NodeMCU  is running OTGW Firmware and tallking to first device via serial =)))) 
and yes, it works )

Running on just a single NodeMCU without need of programming PIC it would be a super nice thing.

rvdbreemen, 2022-11-24:
Cool that it works. Link to your code base. So I can see if I could support it in the future. Thanks 🙏

DaniilMir, 2022-11-24:
> 
Here is the code. I used a default one without any changes.
https://github.com/ihormelnyk/opentherm_library/blob/master/examples/OpenThermGatewayMonitor_Demo/OpenThermGatewayMonitor_Demo.ino

rvdbreemen, 2022-11-24:
@DaniilMir could you explain a bit more what you did? I am not understanding it yet. So let me know.

DaniilMir, 2022-11-24:
@rvdbreemen Sure. 

1. I took one NodeMCU board and flashed it with the "OpenThermGatewayMonitor_Demo.ino"  sketch
2. Connected this board to the DIY OpenTherm adapter based on the [https://github.com/jpraus/arduino-opentherm](url)
3. Took the second NodeMCU board and flashed it with your OTGW Firmware
4. Connected serial ports of both Node MCU's together.
So the first node MCU is providing serial output similar to PIC controller output from original OpenTherm adapter schematic.
And the second one is parsing this output and talking to MQTT server )

rvdbreemen, 2023-01-28:
Okay, that does make sense that my firmware, if the OT stream is similar to that of the OTGW PIC will decode it. This would mean if I integrate that directly I could support other hardware variants.

However that's not been my focus so far, and I have no plans to add this, unless there is a real demand for this. And if it would add value to my project. For now I will only support the hardware variant from NodoShop with the PIC controller, as that's a gateway implementation.

So maybe in the future, who knows.

closed 2023-01-28 by rvdbreemen